TheRealMoeder/Energy2Shelly_ESP #1: Welches Port für Zugang

martind40, 2025-02-17:
Hab nun einen ESP12 (=8266) damit geflashed. Über 192.168.4.1 konnte ich noch meine WIFI Daten eingeben. Seh den Chip auch am Router. Aber kann darauf nicht zugreifen. Weder mit Port 1010 auch nicht mit 7777 oder mit /status.
Wie kann ich einsehen, was der Chip gerade empfängt, oder weiterleitet?

Danke dir.

TheRealMoeder, 2025-02-17:
Aktuell werden die Daten nur direkt an den virtuellen shelly geschickt. Debug Daten werden nur seriell ausgegeben, wenn entsprechend konfiguriert.
Hast du die shelly ID (= MAC Adresse in Uni-Meter) willkürlich vergeben? Wenn ja müsstest du in der Hoymiles App den Shelly hinzufügen können.

TheRealMoeder, 2025-02-17:
Ich habe einen aktualisierten Build hinzugefügt, welcher die unter /status ebenfalls die PowerDaten anzeigt.

TheRealMoeder, 2025-02-19:
@martind40 es gibt einige Verbesserungen, probier die neue Version mal aus.

martind40, 2025-02-20:
mach ich @TheRealMoeder . Komme aber erst morgen dazu.

martind40, 2025-02-21:
Auch hier bekomme ich keine Daten vom SHRDZM. Irgendwo hab ich einen Denkfehler.

{"id":1,"src":"shellypro3em-a4cf12ae25c8","dst":"user_1","result":{"id":0,"a_current":0,"a_voltage":0,"a_act_power":0,"a_aprt_power":0,"a_pf":0,"a_freq":0,"b_current":0,"b_voltage":0,"b_act_power":0,"b_aprt_power":0,"b_pf":0,"b_freq":0,"c_current":0,"c_voltage":0,"c_act_power":0,"c_aprt_power":0,"c_pf":0,"c_freq":0,"total_current":0,"total_act_power":0,"total_aprt_power":0}}

TheRealMoeder, 2025-02-21:
Shrdzm habe ich noch nicht integriert, da ich selber keinen zum testen besitze. Sollte allerdings relativ einfach umzusetzen sein, wenn ich den Code von Sdeigm anschaue...

martind40, 2025-02-22:
Hab mal UDP am SHRDZM aktiviert, wie hier https://github.com/sdeigm/uni-meter/issues/5 beim Uni-Meter beschrieben.
E2S zeigt dennoch nur "0en" bei den Werten an. UDP wird aber abgefragt im Programm.

TheRealMoeder, 2025-02-22:
Er wertet im Moment nur die UDP Telegramme von SMA aus. Ich arbeite an der Integration deines Zählers aber schon...

TheRealMoeder, 2025-02-22:
Bitte Fortführung in den entsprechenden Issues.

closed 2025-02-22 by TheRealMoeder

TheRealMoeder, 2025-02-22:
> Hab mal UDP am SHRDZM aktiviert, wie hier [sdeigm/uni-meter#5](https://github.com/sdeigm/uni-meter/issues/5) beim Uni-Meter beschrieben. E2S zeigt dennoch nur "0en" bei den Werten an. UDP wird aber abgefragt im Programm.

test build verfügbar, siehe #6