TheRealMoeder/Energy2Shelly_ESP #2: Alten Shelly 3EM in Energy2Shelly emulieren

Crusher606, 2025-02-20:
Hallo,

prima Idee den Emulator auf einem ESP laufen zu lassen.

Ich hätte da ein paar Fragen:

1.) Wäre es möglich das Programm so zu erweitern, das man Energy2Shelly mit Daten eines vorhandenen alten Shelly 3EM über die 
http://<IP-Adresse> emulieren könnte? Mit dem "uni-meter" von "sdeigm" geht das über die Konfigurationsdatei und funktioniert auch gut.
Für die Aufgabe einer Emulation wäre es mir aber schon zu schade, dauerhaft einen Raspberry Pi zu opfern.

2.) Ich habe hier einen ioBroker am laufen, der aktuell die Daten meines Shelly 3EM empfängt, weiß aber nicht wie ich das mit Energy2Shelly umsetzen muss?! Habe keine leider große Erfahrung mit MQTT im ioBroker.
Wie bekommt man die Werte des alten 3EM in den Energy2Shelly, wenn ich die MQTT Adresse und Port von dem MQTT Adapter in meinem ioBroker eingetragen habe? 

Ich weiß leider nicht was man hier laut ReadMe machen muss, da ich leider nicht so damit auskenne.
_Power values on the MQTT topic can be either a raw number or at a configurable JSON field using a JSON Path-style syntax, e.g. "energy.data" for {"energy":{"data":mypowervalue}}._

3.) Wenn ich Energy2Shelly auf den ESP8266 geflasht und eingerichtet habe, kann ich über die <IP>/status die Daten sehen.
Aber wenn ich die Stromversorgung aber trenne und wieder herstelle, ist er in meinem Netzwerk nicht mehr vorhanden?
Mit neu flashen und wieder einbinden über den AP läuft er dann wieder.

TheRealMoeder, 2025-02-20:
Ich habe die Hauptfragen mal in zwei neue Issues ausgegliedert. Bezüglich der Mqtt fragen am besten in den entsprechenden Foren nachhören, kenne mich mit ioBroker nicht (mehr) aus.

TheRealMoeder, 2025-02-21:
deinen. 3. Punkt kann ich bei mir nicht nachvollziehen...

martind40, 2025-02-22:
Hab dasselbe Thema von Punkt 3 von Crusher606. Versuch nun mal über ArduinoIDE meine  WiFi Daten direkt zu hinterlegen. Mal schau'n ob ich das hinbekomm.

TheRealMoeder, 2025-02-22:
Wenn du das machst musst du aber den WiFi Manager deaktivieren und auch die restlichen Sachen händisch hinterlegen. Es wäre hilfreich mal den debug output an der seriellen Schnittstelle zu sehen, den nächsten build mache ich wieder mit debug true

TheRealMoeder, 2025-02-22:
Diskussion in entsprechenden Issues

closed 2025-02-22 by TheRealMoeder

fke222, 2025-02-24:
zu ESP8266 nach Power off nicht mehr mit dem Wlan verbunden.
Hatte das gleiche Problem. Mit "anderem" ESP hat es dann funktioniert. Mein ESP war schon ziemlich alt - er hat die Wifi Daten nicht richtig abgespeichert.

TheRealMoeder, 2025-02-24:
Es lag tatsächlich an der Methode der Datenspeicherung im Programm (siehe #3).