TheRealMoeder/Energy2Shelly_ESP #4: Add support for Shelly 3EM

TheRealMoeder, 2025-02-20:


TheRealMoeder, 2025-02-21:
I will try to enable a generic HTTP input as in uni-meter, but this will require some work on the configuration page, otherwise it wouldn't be to user friendly anymore.

055Richard, 2025-02-24:
Http input would be verder nice.

TheRealMoeder, 2025-02-26:
My first version with generic http input is available in the releases!

055Richard, 2025-02-27:
I get the message     
WS_EVT_PING:
WS_EVT_PONG:
not defined

TheRealMoeder, 2025-02-27:
Everything compiles fine for me with platformio and both esp8266 esp32. Please make sure you have the correct library (esp32async/ESPAsyncWebServer@^3.7.1) and try my bin to make sure there is no issue with your compile environment. If it still doesn't work:

Do you have serial debug output for me?

Your configuration looks right, but maybe this long Json response makes some issues, as I implemented the simple HTTP 1.0 to avoid code complexity with chunked transfer encodings. Can you call http.getString and print it so serial so we can make sure the http request gets handled correctly as a first step?

055Richard, 2025-02-28:
It's working there is data in de webpage.
The meter can not be selected in de marstek app.
The b2500 simulator in HA is working.

closed 2025-02-28 by TheRealMoeder

TheRealMoeder, 2025-02-28:
@055Richard Check release 0.4.4, I tried implementing RPC over UDP for Marstek on port 1010. Please report back.

Crusher606, 2025-03-06:
Hallo,
ich habe die 0.4.4 auf einen ESP8266 geflasht.
Habe mich mit dem AP verbunden, mein WLan eingetragen.
Dann habe ich HTTP in das Feld eingetragen und die URL meines Shelly 3EM (http://192.168.179.30) in das Feld darunter. 
Die Werte gespeichert und Energy2Shelly über die IP mit /status aufgerufen.

Es sind keine Daten vom Shelly 3EM im Status ?!

{
  "id": 1,
  "src": "shellypro3em-112233445566",
  "dst": "user_1",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 0,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 0,
    "a_freq": 0,
    "b_current": 0,
    "b_voltage": 0,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 0,
    "b_freq": 0,
    "c_current": 0,
    "c_voltage": 0,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 0,
    "c_freq": 0,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
  }
}

TheRealMoeder, 2025-03-06:
Du musst die korrekte API URL des Shelly angeben, im Falle des 3em also ip-adresse/status
Unter jsonpath für Power dann den entsprechenden pfaf angeben, meines Wissens ist das "total_power"

Crusher606, 2025-03-06:
Ah, ok, das probiere ich morgen noch mal aus.

Crusher606, 2025-03-15:
Rückmeldung zu Version 0.4.5
- sieht gut aus -

bin-Datei auf einen ESP8266 geflasht
AP von Energy2Shelly aufgerufen

WLan und Passwort eingetragen
"HTTP" eingegeben
URL vom Shelly 3EM eingetragen (http://<IP>/status)
Im JSON Path <total_power> eingetragen
Eingaben gespeichert
ESP im Netzwerk aufgerufen (<IP_DES_ESP>/status)

{"id":0,"src":"shellypro3em-112233445566","result":{"id":0,"a_current":0.56,"a_voltage":230,"a_act_power":129.7,"a_aprt_power":129.7,"a_pf":1,"a_freq":50,"b_current":0.56,"b_voltage":230,"b_act_power":129.7,"b_aprt_power":129.7,"b_pf":1,"b_freq":50,"c_current":0.56,"c_voltage":230,"c_act_power":129.7,"c_aprt_power":129.7,"c_pf":1,"c_freq":50,"total_current":1.69,"total_act_power":389.1,"total_aprt_power":389.1}}

funktioniert =)