TheRealMoeder/Energy2Shelly_ESP #6: Add support for SHRDZM

TheRealMoeder, 2025-02-22:
Moved to separate issue from #1

TheRealMoeder, 2025-02-22:
Please try this first test build for SHRDZM - in theory it should work...

[firmware.bin.zip](https://github.com/user-attachments/files/18924103/firmware.bin.zip)

martind40, 2025-02-22:
Thx. Works via UDP.

{"id":1,"src":"shellypro3em-a4cf12ae25c8","dst":"user_1","result":{"id":0,"a_current":0.38,"a_voltage":230,"a_act_power":87.32,"a_aprt_power":87.32,"a_pf":1,"a_freq":50,"b_current":0.38,"b_voltage":230,"b_act_power":87.32,"b_aprt_power":87.32,"b_pf":1,"b_freq":50,"c_current":0.38,"c_voltage":230,"c_act_power":87.32,"c_aprt_power":87.32,"c_pf":1,"c_freq":50,"total_current":1.138956522,"total_act_power":261.96,"total_aprt_power":261.96}}

TheRealMoeder, 2025-02-22:
Fantastic!

closed 2025-02-22 by TheRealMoeder