TheRealMoeder/Energy2Shelly_ESP #11: Add UDP output for Marstek Venus

TheRealMoeder, 2025-02-23:
It seems the Marstek Venus requires RPC over UDP

Crusher606, 2025-02-24:
Als Info:
Soweit ist weiß ist hier der Port 1010 zu nutzen, damit ein Marstek Venus einen Shelly Pro 3EM findet.

Wird das dann ein zusätzliches Eingabefeld auf der Konfigurationsseite?

martind40, 2025-02-24:
Steht auch so in der MARSTEK Einrichtungshilfe für den Shelly:
https://cdn.shopify.com/s/files/1/0746/1575/0952/files/Instructions_for_using_Venus_with_Shelly_Pro_3EM-V1.0.pdf?v=1737427865

TheRealMoeder, 2025-02-28:
Bitte mal v0.4.4 probieren, kann es selber leider nicht testen. UDP auf 1010 wird automatisch immer beantwortet.

055Richard, 2025-02-28:
Test with marstek venus. still no connection.
Used v0.4.4.

The B2500 software in HA is working.

TheRealMoeder, 2025-02-28:
Any debug output on serial monitor?

055Richard, 2025-02-28:
I don't have a serial port interface. 
Fash the board with usb-cable.

055Richard, 2025-02-28:
17:23:00.199 -> �1��4)(��!%�18��X
17:23:02.255 -> ݁�L��!�1��x����D)��.!�.�W�!)�ѡ)!�)��ʥ
17:23:04.152 -> eI��I��
�!�
!!M!��H!!H!!�
!��
!��J!!��!!�%I!T�!!T��J�J!��!]�
17:23:04.186 -> ��E!�!��)!�!)!!ů��1
J)!�!�K�!QK�!�)�)��)�I!�I�����dI!���)!�!�]H�!9�H!�� !��ͤ�
�!�K!�q
17:23:04.286 -> !�쵥!!�N!!Ř��)!!!���)�+)Q�)!Q
!����J�))�
))��9H!)ޡ�)�)!��!!M�)�pJ!)�!!�)!�,�!���!d��!��6)):�-��H)�
!)��
17:23:04.444 -> �ķ���������(�D)9;q�M�I,lE0jH�)c����� ����ʇ��911�� 
17:23:07.603 -> �ش
17:23:07.603 -> 1����!��1H����!����F)
17:23:09.356 -> �?!X���)��)��Y��^!��ԉ)UB!)��ɉtM!��!���)�)��!
Nd�!�!1
T!��!!��!!�
JL��
17:23:09.469 -> ��!!]*���){)э!
!)�))�)
�ki�
17:23:09.515 -> !���!�h1)�)�̮�!�W����9)�
17:23:09.515 -> �!�K)q�!!iK�!�
|��))��M
17:23:09.550 -> �
�!�I!��
)!�ͤ)�!�1�-!��"!��ͬ�
*!1�!�o!�!��)�)

TheRealMoeder, 2025-02-28:
set the serial monitor on usb to 115200 baud please

055Richard, 2025-02-28:
17:28:54.168 -> *wm:Connecting to SAVED AP: Bla
17:28:54.655 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
17:28:55.494 -> *wm:AutoConnect: SUCCESS 
17:28:55.494 -> *wm:STA IP Address: 192.168.178.166
17:28:55.494 -> connected
17:28:55.494 -> The values in the preferences are: 
17:28:55.494 -> 	input_type : HTTP
17:28:55.494 -> 	mqtt_server : http://192.168.178.144/api/v1/data
17:28:55.494 -> 	mqtt_port : 
17:28:55.494 -> 	mqtt_topic : 
17:28:55.494 -> 	mqtt_user : 
17:28:55.494 -> 	mqtt_passwd : 
17:28:55.494 -> 	power_path : active_power_w
17:28:55.494 -> 	energy_in_path : 
17:28:55.494 -> 	energy_out_path : 
17:28:55.494 -> 	query_period : 1000
17:28:55.494 -> 	shelly_mac : 485519ecd7c8
17:28:55.494 -> Enabling generic HTTP data input
17:28:55.531 -> local ip
17:28:55.531 -> 192.168.178.166
17:28:55.531 -> mDNS responder started
17:28:56.305 -> 
17:28:56.305 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:28:56.305 -> 
17:28:56.305 -> Exception (28):
17:28:56.305 -> epc1=0x40232682 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
17:28:56.305 -> 
17:28:56.305 -> >>>stack>>>
17:28:56.305 -> 
17:28:56.305 -> ctx: cont
17:28:56.535 -> 3ffffe30:  401005b2 00000030 00000010 ffffffff  
17:28:56.535 -> 3ffffe40:  401005a4 00000000 4bc6a7f0 00000025  
17:28:56.535 -> 3ffffe50:  00000041 4f511fe2 00000000 fffffffe  
17:28:56.535 -> 3ffffe60:  00000026 3fffc6fc 26b01fe2 00000000  
17:28:56.567 -> 3ffffe70:  0000098b 4f5c28f5 000b0913 00000030  
17:28:56.567 -> 3ffffe80:  40211484 3fffc6fc 00000001 3fff0448  
17:28:56.567 -> 3ffffe90:  3fff0448 3fff041c 3fff06b0 00000030  
17:28:56.567 -> 3ffffea0:  4000050c 3fffc200 00000022 3fffff30  
17:28:56.567 -> 3ffffeb0:  40227ff0 00000030 00000010 ffffffff  
17:28:56.567 -> 3ffffec0:  40227fed 00000050 3ff00014 00000000  
17:28:56.567 -> 3ffffed0:  00000000 00000000 00000000 fffffffe  
17:28:56.599 -> 3ffffee0:  ffffffff 3fffc6fc 00000001 3fffdab0  
17:28:56.599 -> 3ffffef0:  00000000 3fffdad0 3fff06b0 00000030  
17:28:56.599 -> 3fffff00:  00000000 00000000 00000001 3fffdab0  
17:28:56.599 -> 3fffff10:  00000000 3fffdad0 3fff06b0 00000030  
17:28:56.599 -> 3fffff20:  00000000 0000098b ad916872 000b0980  
17:28:56.599 -> 3fffff30:  3fff0000 00000000 00000000 3fff1a3c  
17:28:56.599 -> 3fffff40:  3ffe8650 00000000 3fff283c 00080008  
17:28:56.631 -> 3fffff50:  3fff23fc 00000001 00000001 402113a2  
17:28:56.631 -> 3fffff60:  00000000 00000000 3ffe8670 3fffff4c  
17:28:56.631 -> 3fffff70:  00040001 0000ffff 00050000 ffff0620  
17:28:56.631 -> 3fffff80:  3fffff40 3ffe8ecb 00000003 3fff0601  
17:28:56.631 -> 3fffff90:  3fffdad0 00000000 3fff0684 3fff06b0  
17:28:56.631 -> 3fffffa0:  3fffdad0 00000000 3fff0684 4020b683  
17:28:56.631 -> 3fffffb0:  00000000 00000000 3fff0684 40228270  
17:28:56.664 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 401003b1  
17:28:56.664 -> <<<stack<<<
17:28:56.664 -> 
17:28:56.664 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:28:56.664 -> 
17:28:56.664 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
17:28:56.664 -> 
17:28:56.664 -> load 0x4010f000, len 3424, room 16 
17:28:56.700 -> tail 0
17:28:56.700 -> chksum 0x2e
17:28:56.700 -> load 0x3fff20b8, len 40, room 8 
17:28:56.700 -> tail 0
17:28:56.700 -> chksum 0x2b
17:28:56.700 -> csum 0x2b
17:28:56.700 -> v00074190
17:28:56.700 -> ~ld

055Richard, 2025-02-28:
Is this helping?

TheRealMoeder, 2025-02-28:
[firmware.bin.zip](https://github.com/user-attachments/files/19030744/firmware.bin.zip)

please try this debugging build and report the serial output again.

055Richard, 2025-02-28:
I can only install the main.cpp the arduino ide software

TheRealMoeder, 2025-02-28:
[main.cpp.zip](https://github.com/user-attachments/files/19031265/main.cpp.zip)

055Richard, 2025-02-28:
18:38:10.903 -> tail 0
18:38:10.903 -> chksum 0x2b
18:38:10.903 -> csum 0x2b
18:38:10.903 -> v000741a0
18:38:10.943 -> ~ld
18:38:10.977 -> ����o�s��g|�$�$`c��|r�l�g��g�$`��{�l�$��*wm:AutoConnect 
18:38:11.225 -> *wm:Connecting to SAVED AP: Wolbrink
18:38:11.703 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
18:38:14.026 -> *wm:AutoConnect: SUCCESS 
18:38:14.026 -> *wm:STA IP Address: 192.168.178.166
18:38:14.026 -> connected
18:38:14.026 -> The values in the preferences are: 
18:38:14.026 -> 	input_type : HTTP
18:38:14.060 -> 	mqtt_server : http://192.168.178.144/api/v1/data
18:38:14.060 -> 	mqtt_port : 
18:38:14.060 -> 	mqtt_topic : 
18:38:14.060 -> 	mqtt_user : 
18:38:14.060 -> 	mqtt_passwd : 
18:38:14.060 -> 	power_path : active_power_w
18:38:14.060 -> 	energy_in_path : 
18:38:14.060 -> 	energy_out_path : 
18:38:14.060 -> 	query_period : 1000
18:38:14.060 -> 	shelly_mac : 485519ecd7c8
18:38:14.060 -> Enabling generic HTTP data input
18:38:14.060 -> local ip
18:38:14.060 -> 192.168.178.166
18:38:14.060 -> mDNS responder started
18:38:15.081 -> Querying HTTP source
18:38:15.111 -> Current total power: 1189.00
18:38:15.112 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:38:16.079 -> Querying HTTP source
18:38:16.116 -> Current total power: 1174.00
18:38:16.116 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:38:17.056 -> {"id":0,"method":"EM.GetStatus","params":{"id":0}}
18:38:17.056 -> 
18:38:17.056 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:38:17.056 -> 
18:38:17.056 -> Exception (28):
18:38:17.056 -> epc1=0x4023268a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
18:38:17.087 -> 
18:38:17.087 -> >>>stack>>>
18:38:17.087 -> 
18:38:17.087 -> ctx: cont
18:38:17.087 -> sp: 3ffff9f0 end: 3fffffd0 offset: 0150

TheRealMoeder, 2025-02-28:
[main.cpp.zip](https://github.com/user-attachments/files/19031509/main.cpp.zip)

Ah, ok, stupid me. Next try.

TheRealMoeder, 2025-02-28:
It's on my agenda, but I have to rework the configuration page, it's becoming very user unfriendly with all those new configuration options.

055Richard, 2025-02-28:
I think that the connection is not that stable. Th Marstek starts every 2-3 loading 2000kw without a reason.
When yoy switch a meter of is comming online this happens also.

18:59:39.861 -> Current total power: 1547.00
18:59:39.861 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:40.591 -> Received UDP packet on port 1010: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
18:59:40.591 -> {"id":0,"src":"shellypro3em-485519ecd7c8","dst":"user_1","result":{"id":0,"a_current":2.24,"a_voltage":230,"a_act_power":515.62,"a_aprt_power":515.62,"a_pf":1,"a_freq":50,"b_current":2.24,"b_voltage":230,"b_act_power":515.62,"b_aprt_power":515.62,"b_pf":1,"b_freq":50,"c_current":2.24,"c_voltage":230,"c_act_power":515.62,"c_aprt_power":515.62,"c_pf":1,"c_freq":50,"total_current":6.725478261,"total_act_power":1546.86,"total_aprt_power":1546.86}}
18:59:40.744 -> Querying HTTP source
18:59:40.823 -> Current total power: 1535.00
18:59:40.823 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:41.753 -> Querying HTTP source
18:59:41.798 -> Current total power: 1550.00
18:59:41.798 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:42.405 -> Received UDP packet on port 1010: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
18:59:42.405 -> {"id":0,"src":"shellypro3em-485519ecd7c8","dst":"user_1","result":{"id":0,"a_current":2.25,"a_voltage":230,"a_act_power":516.62,"a_aprt_power":516.62,"a_pf":1,"a_freq":50,"b_current":2.25,"b_voltage":230,"b_act_power":516.62,"b_aprt_power":516.62,"b_pf":1,"b_freq":50,"c_current":2.25,"c_voltage":230,"c_act_power":516.62,"c_aprt_power":516.62,"c_pf":1,"c_freq":50,"total_current":6.738521739,"total_act_power":1549.86,"total_aprt_power":1549.86}}
18:59:42.750 -> Querying HTTP source
18:59:42.783 -> Current total power: 1547.00
18:59:42.783 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:43.768 -> Querying HTTP source
18:59:43.804 -> Current total power: 1604.00
18:59:43.804 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:44.220 -> Received UDP packet on port 1010: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
18:59:44.220 -> {"id":0,"src":"shellypro3em-485519ecd7c8","dst":"user_1","result":{"id":0,"a_current":2.32,"a_voltage":230,"a_act_power":534.61,"a_aprt_power":534.61,"a_pf":1,"a_freq":50,"b_current":2.32,"b_voltage":230,"b_act_power":534.61,"b_aprt_power":534.61,"b_pf":1,"b_freq":50,"c_current":2.32,"c_voltage":230,"c_act_power":534.61,"c_aprt_power":534.61,"c_pf":1,"c_freq":50,"total_current":6.973173913,"total_act_power":1603.83,"total_aprt_power":1603.83}}
18:59:44.729 -> Querying HTTP source
18:59:44.874 -> Current total power: 1576.00
18:59:44.874 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:45.744 -> Querying HTTP source
18:59:46.047 -> Current total power: 2450.00
18:59:46.047 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:46.729 -> Querying HTTP source
18:59:46.806 -> Current total power: 3652.00
18:59:46.806 -> Total consumption: 0.00 - Total Grid Feed-In: 0.00
18:59:47.357 -> Received UDP packet on port 1010: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
18:59:47.357 -> {"id":0,"src":"shellypro3em-485519ecd7c8","dst":"user_1","result":{"id":0,"a_current":5.29,"a_voltage":230,"a_act_power":1217.21,"a_aprt_power":1217.21,"a_pf":1,"a_freq":50,"b_current":5.29,"b_voltage":230,"b_act_power":1217.21,"b_aprt_power":1217.21,"b_pf":1,"b_freq":50,"c_current":5.29,"c_voltage":230,"c_act_power":1217.21,"c_aprt_power":1217.21,"c_pf":1,"c_freq":50,"total_current":15.87665217,"total_act_power":3651.63,"total_aprt_power":3651.63}}
18:59:47.737 -> Querying HTTP source

TheRealMoeder, 2025-02-28:
Ok, this will be a little trial and error. Try this one.

[main.cpp.zip](https://github.com/user-attachments/files/19032036/main.cpp.zip)

055Richard, 2025-03-02:
It's not working with de Marstek Venus. Power is -3000, so delivering power to the grid and the Marstek Venus is not loading. 
When I switch to the B2500 HA, then the Marstek is load max, what is expected. 

I stay with the HA B2500 solution.

Succes with the development.

Crusher606, 2025-03-15:
Ist hier etwas zu beachten wenn ich das bin-File auf meinem ESP32 teste?
Kann ich nach dem Flashen und definieren des Shelly 3EM über die URL 

http://<IP_DES_ESP32>/rpc/EM.GetStatus?id=0

zum Testen auf die Daten zugreifen?

TheRealMoeder, 2025-03-15:
Ich würde für weitere Tests das offizielle 0.4.5 Release empfehlen. Zum testen können die Daten direkt über ipdesesp32/status abgerufen werden.

TheRealMoeder, 2025-03-18:
@Crusher606 Zugriff über http://<IP_DES_ESP32>/rpc/EM.GetStatus geht mit release 0.4.6

TheRealMoeder, 2025-03-20:
Leider kann ich RPC over UDP für den Marstek Venus nicht testen, wenn jemand dafür Zeit hat bitte melden!

Crusher606, 2025-03-23:
Heute getestet:
Energy2Shelly_ESP Version 0.4.6

Ich habe die *.bin geflasht
Im AP folgende Daten eingegeben:

WLan mit Passwort
HTTP
http://192.168.179.30/status       (mein Shelly 3EM)
TRIPHASE

Ich sehe diese Daten unter der IPdesESP/status aber die Werte vom Shelly fehlen darin.

{
  "id": 0,
  "src": "shellypro3em-349454816898",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 230,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0,
    "b_voltage": 230,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0,
    "c_voltage": 230,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
  }
}

unter http://http://IPdesESP/rpc/EM.GetStatus sehe ich auch Daten ohne Werte:

{
  "id": 0,
  "src": "shellypro3em-349454816898",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 230,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0,
    "b_voltage": 230,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0,
    "c_voltage": 230,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
[... 219 lines cut ...]
[15:48:00]Querying HTTP source
[15:48:00]resolving triphase
[15:48:00]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:01]Querying HTTP source
[15:48:01]resolving triphase
[15:48:01]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:02]Querying HTTP source
[15:48:02]resolving triphase
[15:48:02]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:03]Querying HTTP source
[15:48:03]resolving triphase
[15:48:03]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:04]Querying HTTP source
[15:48:04]resolving triphase
[15:48:04]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:04]Received UDP packet on port 1010: {"id":0,"method":"EM.GetStatus","params":{"id":0}}
[15:48:04]{"id":0,"src":"shellypro3em-349454816898","result":{"id":0,"a_current":0,"a_voltage":230,"a_act_power":0,"a_aprt_power":0,"a_pf":1,"a_freq":50,"b_current":0,"b_voltage":230,"b_act_power":0,"b_aprt_power":0,"b_pf":1,"b_freq":50,"c_current":0,"c_voltage":230,"c_act_power":0,"c_aprt_power":0,"c_pf":1,"c_freq":50,"total_current":0,"total_act_power":0,"total_aprt_power":0}}
[15:48:05]Querying HTTP source
[15:48:05]resolving triphase
[15:48:05]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:06]Querying HTTP source
[15:48:06]resolving triphase
[15:48:06]Current power L1: 0.00 - L2: 0.00 - L3: 0.00
[15:48:07]Querying HTTP source
```

TheRealMoeder, 2025-03-23:
Du musst dem ESP noch den JSON Pfad zu den Power-Daten geben. Im Falle eines Shelly 3EM schreibe mal anstatt von `TRIPHASE` in das gleiche Feld `total_power`

Crusher606, 2025-03-23:
OK, geändert.

{
  "id": 1,
  "src": "shellypro3em-349454816898",
  "dst": "user_1",
  "result": {
    "id": 0,
    "a_current": 0.01,
    "a_voltage": 230,
    "a_act_power": 1.6,
    "a_aprt_power": 1.6,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0.01,
    "b_voltage": 230,
    "b_act_power": 1.6,
    "b_aprt_power": 1.6,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0.01,
    "c_voltage": 230,
    "c_act_power": 1.6,
    "c_aprt_power": 1.6,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0.02,
    "total_act_power": 4.8,
    "total_aprt_power": 4.8
  }
}

Es sind aber für den Marstek Venus die drei einzelnen Phasen erforderlich.
Diese Daten bekomme ich über das uni-meter, mit dem mein Venus E zur Zeit regelt:

{
  "id": 0,
  "a_current": 1.72,
  "a_voltage": 233.32,
  "a_act_power": -195.22,
  "a_aprt_power": 398.408163265306,
  "a_pf": -0.49,
  "a_freq": 50,
  "b_current": 0.88,
  "b_voltage": 232.84,
  "b_act_power": 114.3,
  "b_aprt_power": 204.107142857143,
  "b_pf": 0.56,
  "b_freq": 50,
  "c_current": 0.66,
  "c_voltage": 233.55,
  "c_act_power": 99.03,
  "c_aprt_power": 154.734375,
  "c_pf": 0.64,
  "c_freq": 50,
  "total_current": 3.26,
  "total_act_power": 18.11,
  "total_aprt_power": 757.25
}

Crusher606, 2025-03-23:
Diese Daten kommen über die Abfrage über den Status vom 3EM rein:
Da sind die einzelnen Phasen bei _emeters_ zu sehen
{
  "wifi_sta": {
    "connected": true,
    "ssid": "Mein-Netzwerk",
    "ip": "192.168.179.30",
    "rssi": -32
  },
  "cloud": {
    "enabled": false,
    "connected": false
  },
  "mqtt": {
    "connected": true
  },
  "time": "16:41",
  "unixtime": 1742744515,
  "serial": 1119,
  "has_update": false,
  "mac": "112233445566",
  "cfg_changed_cnt": 0,
  "actions_stats": {
    "skipped": 0
  },
  "relays": [
    {
      "ison": false,
      "has_timer": false,
      "timer_started": 0,
      "timer_duration": 0,
      "timer_remaining": 0,
      "overpower": false,
      "is_valid": true,
      "source": "input"
    }
  ],
  "emeters": [
    {
      "power": -149.99,
      "pf": -0.44,
      "current": 1.48,
      "voltage": 233.41,
      "is_valid": true,
      "total": 3809672.2,
      "total_returned": 641831.3
    },
    {
      "power": 103.99,
      "pf": 0.54,
      "current": 0.84,
      "voltage": 232.17,
      "is_valid": true,
      "total": 3498721.8,
      "total_returned": 359340.8
    },
    {
      "power": 41.77,
      "pf": 0.34,
      "current": 0.53,
      "voltage": 233.58,
      "is_valid": true,
      "total": 1897315.8,
      "total_returned": 646425
    }
  ],
  "total_power": -4.23,
  "emeter_n": {
    "current": 0,
    "ixsum": 1.78,
    "mismatch": false,
    "is_valid": false
  },
  "fs_mounted": true,
  "v_data": 1,
  "ct_calst": 0,
  "update": {
    "status": "idle",
    "has_update": false,
    "new_version": "20230913-114244/v1.14.0-gcb84623",
    "old_version": "20230913-114244/v1.14.0-gcb84623",
    "beta_version": "20231107-165007/v1.14.1-rc1-g0617c15"
  },
  "ram_total": 49920,
  "ram_free": 28452,
  "fs_size": 233681,
  "fs_free": 154365,
  "uptime": 2345
}

TheRealMoeder, 2025-03-23:
> Es sind aber für den Marstek Venus die drei einzelnen Phasen erforderlich. Diese Daten bekomme ich über das uni-meter, mit dem mein Venus E zur Zeit regelt:

Ärgerlich. Dann muss ich leider mein JSON-Path handling überarbeiten, da ich aktuell nicht auf das emeters-Array zugreifen kann...
Etwas Geduld bitte.

Crusher606, 2025-03-23:
Kein Stress . . . . .

Crusher606, 2025-03-23:
Diese Daten bekomme ich, wenn ich das uni-meter Abfrage, das auf meinem Rasperry Pi läuft und mit dem Venus E gekoppelt ist.

http://IP_DES_RPi3_uni-meter/rpc/EM.GetStatus?id=0

{"id":0,"a_current":0.83,"a_voltage":232.24,"a_act_power":-130.82,"a_aprt_power":189.59420289855072,"a_pf":-0.69,"a_freq":50.0,"b_current":1.02,"b_voltage":233.73,"b_act_power":164.71,"b_aprt_power":238.71014492753625,"b_pf":0.69,"b_freq":50.0,"c_current":0.46,"c_voltage":232.89,"c_act_power":-33.01,"c_aprt_power":106.48387096774194,"c_pf":-0.31,"c_freq":50.0,"total_current":2.31,"total_act_power":0.88,"total_aprt_power":534.79}

TheRealMoeder, 2025-03-24:
[firmware.bin.zip](https://github.com/user-attachments/files/19429534/firmware.bin.zip)

Hier ein Testbuild aus dem Branch jsonpath_full für ESP8266. Das ist noch keine volle JSON Path Implementation, genügt aber um auf das emeters-Array des 3EM zuzugreifen. Bitte mal testen mit `TRIPHASE` und dann für die einzelnen Phasenleistungen:
`emeters[0].power`, `emeters[1].power` und `emeters[2].power`

Crusher606, 2025-03-28:
Eben getestet.
Energy2Shelly_ESP Version aus dem Test_zip File

Ich habe die *.bin geflasht
Im AP folgende Daten eingegeben:

WLan mit Passwort
HTTP
http://192.168.179.30/status (mein Shelly 3EM)
TRIPHASE
emeters[0].power
emeters[1].power
emeters[2].power

Das steht im Shelly 3EM Webinterface:
-------------------------------------------
{
  "wifi_sta": {
    "connected": true,
    "ssid": "MEIN_NETZTWERK",
    "ip": "VOM_SHELLY_3EM",
    "rssi": -38
  },
  "cloud": {
    "enabled": false,
    "connected": false
  },
  "mqtt": {
    "connected": true
  },
  "time": "16:56",
  "unixtime": 1743177415,
  "serial": 61,
  "has_update": false,
  "mac": "112233445566",
  "cfg_changed_cnt": 0,
  "actions_stats": {
    "skipped": 0
  },
  "relays": [
    {
      "ison": false,
      "has_timer": false,
      "timer_started": 0,
      "timer_duration": 0,
      "timer_remaining": 0,
      "overpower": false,
      "is_valid": true,
      "source": "input"
    }
  ],
  "emeters": [
    {
      "power": -76.87,
      "pf": -0.28,
      "current": 1.19,
      "voltage": 234.81,
      "is_valid": true,
      "total": 3826567.3,
      "total_returned": 648827.2
    },
    {
      "power": 66.2,
      "pf": 0.42,
      "current": 0.68,
      "voltage": 234.71,
      "is_valid": true,
      "total": 3513656.3,
      "total_returned": 367193.5
    },
    {
      "power": -4.39,
      "pf": -0.03,
      "current": 0.55,
      "voltage": 233.73,
      "is_valid": true,
      "total": 1900922,
      "total_returned": 655497.8
    }
  ],
  "total_power": -15.06,
  "emeter_n": {
    "current": 0,
    "ixsum": 0.97,
    "mismatch": false,
    "is_valid": false
  },
  "fs_mounted": true,
  "v_data": 1,
  "ct_calst": 0,
  "update": {
    "status": "idle",
    "has_update": false,
    "new_version": "20230913-114244/v1.14.0-gcb84623",
    "old_version": "20230913-114244/v1.14.0-gcb84623",
    "beta_version": "20231107-165007/v1.14.1-rc1-g0617c15"
  },
  "ram_total": 49920,
  "ram_free": 28396,
  "fs_size": 233681,
  "fs_free": 154365,
  "uptime": 107
}

Webinterface vom ESP8266 aufgerufen: (bei mir http://192.168.179.117/status)
---------------------------------------------------------------------------------------
{
  "id": 0,
  "a_current": -0.22,
  "a_voltage": 230,
  "a_act_power": -53.08,
  "a_aprt_power": -53.08,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": 0.29,
  "b_voltage": 230,
  "b_act_power": 65.65,
  "b_aprt_power": 65.65,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": 0.1,
  "c_voltage": 230,
  "c_act_power": 23.04,
  "c_aprt_power": 23.04,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": 0.15,
  "total_act_power": 35.61,
  "total_aprt_power": 35.61
}

Webinterface vom ESP8266 aufgerufen: (http://192.168.179.117/rpc/EM.GetStatus?id=0)
-------------------------------------------------------------------------------------------------
{
  "id": 0,
  "a_current": -0.48,
  "a_voltage": 230,
  "a_act_power": -113.03,
  "a_aprt_power": -113.03,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": 0.29,
  "b_voltage": 230,
  "b_act_power": 67.05,
  "b_aprt_power": 67.05,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": 0.13,
  "c_voltage": 230,
  "c_act_power": 29.4,
  "c_aprt_power": 29.4,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": -0.06,
  "total_act_power": -16.58,
  "total_aprt_power": -16.58
}

Crusher606, 2025-03-28:
Ich muss aber noch testen, ob der Marstek Venus E das Signal vom ESP8266 als Shelly Pro 3EM erkannt und nutzen kann.

TheRealMoeder, 2025-03-29:
Das sieht doche erstmal gut aus. Ich würde sagen ein Test mit dem Venus steht nichts entgegen.

Crusher606, 2025-03-29:
Test war erfolgreich. 

Marstek Venus E hat den emulierten Shelly PRO 3EM auf dem ESP8266 gefunden.
Die Regelung wird jede Sekunde angepasst. (1000 ms)

Im Marstek Venus E Webinterface sieht man die Regelkurven.
Darunter stehen die Werte für: 
Total:
Phase A:
Phase B:
Phase C:
Venus E:

Wenn das Programm auch noch auf einem etwas schnelleren ESP32 läuft wäre das perfekt.
(Leider sehe ich nicht die Auslastung des ESP8266)

Crusher606, 2025-03-29:
Ich lasse den Venus E jetzt an dem ESP8266 mit  Energy2Shelly gekoppelt. Mal sehen ob es morgen im Laufe des Tage stabil läuft oder ob etwas komisches passiert.

TheRealMoeder, 2025-03-29:
Super. esp32 läuft, aber der 8266 genügt völlig

Crusher606, 2025-03-30:
Energy2Shelly funktioniert.
Der ESP8266 sendet die Shelly 3EM Daten an den Venus E und der regelt damit.
Ich lasse es jetzt so laufen und beobachte weiter.

TheRealMoeder, 2025-03-30:
Super, das freut micht. Wie gesagt ESP32 lässt sich einwandfrei kompilieren, die bins lade ich mit dem nächsten Release hoch, aber die sekundlichen abfragen des Venus kann der esp8266 problemlos beantworten, sodass mehr Geschwindigkeit nichts bringt.

closed 2025-03-30 by TheRealMoeder