TheRealMoeder/Energy2Shelly_ESP #8: Move to different websocket library to make compilation for esp32 possiblr

TheRealMoeder, 2025-02-23:
https://github.com/ESP32Async/ESPAsyncWebServer

TheRealMoeder, 2025-02-24:
[Energy2Shelly_v0_4_2_ESP32.zip](https://github.com/user-attachments/files/18950953/Energy2Shelly_v0_4_2_ESP32.zip)

TheRealMoeder, 2025-02-24:
In theory it should work, the only thing I'm not sure about is the mDNS broadcast

TheRealMoeder, 2025-02-26:
[Energy2Shelly_v0_4_3_ESP32.zip](https://github.com/user-attachments/files/18993809/Energy2Shelly_v0_4_3_ESP32.zip)

Updated ESP32 test build for version 0.4.3

martind40, 2025-02-28:
Lässt sich zwar hochladen, aber es öffnet sich nicht der WiFiManager. Aber auch mit den Versionen 0.4.2 & 0.4.4 für den ESP32 nicht.

TheRealMoeder, 2025-02-28:
Wahrscheinlich weil die WiFi Daten noch gespeichert sind und er sich direkt mit dem AP verbindet.

martind40, 2025-02-28:
Ich lösche die ESPs vor dem flashen immer, daher nicht nachvollziehbar woher er die WIFI Daten haben soll.
Ja, bootloader.bin hab ich mit geladen. Nur ladet er verdammt schnell. Eher ungewöhnlich.

TheRealMoeder, 2025-02-28:
[esp32.zip](https://github.com/user-attachments/files/19032825/esp32.zip)

bitte nochmal probieren...

martind40, 2025-03-01:
Nein leider auch nicht. Noch immer kein WIFI. 
Nur irgendwas muss sich am Programm geändert haben, da das Programm nun "normal" lange braucht um hochgeladen zu werden.

TheRealMoeder, 2025-03-01:
Solange ich keine eigenen esp32 habe wird die Arbeit hier erstmal ruhen, da ich es sonst nicht sinnvoll testen kann. Habe im aktuellen Main branch auch noch zwei Probleme die ich zunächst beheben muss

TheRealMoeder, 2025-03-18:
@martind40 Ich habe beim ESP32 release die partitions.bin vergessen, im Release 0.4.6 ist sie auch enthalten. Bitte damit erneut probieren.

TheRealMoeder, 2025-03-20:
Meine ESP32 sind angekommen und ich kann bestätigen, das die Funktionalität auf ESP32 und ESP8266 identisch ist, wenn korrekt geflasht wurde

closed 2025-03-20 by TheRealMoeder