TheRealMoeder/Energy2Shelly_ESP #13: Wie Werte mit mqtt senden

fke222, 2025-02-25:
Hallo,
die akt. Version (v0.4.2) hat sich für einen ESP32 kompilieren lassen und startet auch wie gewollt.
Ich konnte ihn sogar im Hoymiles "sehen" das Hizufügen klappte aber nicht - habe dafür keine Fehlermeldung gesehen / gefunden.
Könntest du bitte noch ein Beispiel nennen, wie ich auf der Cl mit mosqitto_pub Werte Übergeben kann. Bin leider kein Profi was JSON betrifft.
Wie sollten funktionierende Eingaben sein für:
Energy.POWER
.....
mosquitto_pub -t 'tele/meter/SENSOR' -m  {"Energy":{"Power":200}} ????

Danke schon mal vorab für die super Arbeit.

TheRealMoeder, 2025-02-25:
Das du den ESP in der Hoymiles App sehen kannst ist schon mal super. Wahrscheinlich ist aber der mDNS nicht vollständig und er verweigert deshalb das hinzufügen...kannst du auf dem serial debug output etwa sehen?

Zu mosquitto_pub: es fehlt noch die Angabe des MQTT Servers mit -h und um die Json Daten mit -m gehören noch Anführungszeichen, deshalb musst du die eigentlichen Anführungszeichen escapen:
mosquitto_pub -h IPdesMQTTServer -t tele/Meter/Sensor -m "{\"Energy\":{\"Power\":200}"

fke222, 2025-02-26:
Danke für die schnelle Antwort.
Werde in den nächsten Tagen mal mit dem Seriellen Monitor den Verbindungsversuch überwachen.