TheRealMoeder/Energy2Shelly_ESP #15: Fehler beim Compilieren "Compilation error: 'WS_EVT_PING' was not declared in this scope"

fke222, 2025-02-27:
ESP32 DevModule:
beim Compilieren kommt es zu dieser Fehlermeldung:
Compilation error: 'WS_EVT_PING' was not declared in this scope,
wenn ich case WS_EVT_PING: auskommentiere kann ich den Sketch hochladen.

Nach dem Connect ins WLAN und Verbindung zum MQTT Broker erscheint Shelly Pro em3 In der Hoymiles App.
Beim Versuch dies hinzuzufügen kommt der Fehler in der Hoymiles App "kann Daten nicht laden". Beim Versuch den "Shelly" manuel in der Hoymiles App hinzuzufügen kommt ebenfalls eine Fehlermeldung - kann "Device nicht finden" (oder so ähnlich - wird nur kurz angezeigt).
Statusseite  ist über Wlan erreichbar. 
Im seriellen Monitor kommen keine Meldungen.

TheRealMoeder, 2025-02-27:
Hast du den build aus #8 mal ausprobiert? Ich muss anscheinend doch ein paar esp32 zu Testzwecken bestellen...

TheRealMoeder, 2025-02-28:
Wahrscheinlich falsche library installiert. bitte entsprechend der angegebenen libs in der platformio.ini kompilieren, dann geht es auch ohne Fehlermeldungen.

closed 2025-02-28 by TheRealMoeder

fke222, 2025-03-01:
Hi,
Vielen Dank,
War die falsche Lib.
Nun funktioniert es.
Danke
Fritz

<https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/15#issuecomment-2
DCZPAVCNFSM6AAAAABYAD6UDSVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMOJRGUYTC
MZQGU> .

<https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/15#issuecomment-2
691511305> 

<https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/15#issuecomment-2
DCZPAVCNFSM6AAAAABYAD6UDSVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMOJRGUYTC
MZQGU> .
[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
"https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/15#issuecomment-2
691511305", "url": 
"https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/15#issuecomment-2
691511305", "name": "View Issue" }, "description": "View this Issue on
"https://github.com" } } ]