TheRealMoeder/Energy2Shelly_ESP #16: Add MQTT-User and Password

Higgy72, 2025-02-28:
Bitte die Möglichkeit zur Eingabe von MQTT-Nutzer und -Passwort ergänzen!
Danke!

TheRealMoeder, 2025-02-28:
Please try release 0.4.4

martind40, 2025-02-28:
0.4.4 (esp8266) funtioniert so weit; Für den ESP32 öffnet sich nach dem Flashen aber der WiFiManager nicht (also kein AP). Habe eben drei verschiedene ESP32 getestet.

TheRealMoeder, 2025-02-28:
MQTT Anmeldung klappt auch? Dann können wir hier zu machen.

Higgy72, 2025-02-28:
Vielen Dank für die schnelle Umsetzung!
Habe leider Keine Ahnung wie ich die korrekte Anmeldung kontrollieren kann. Jedenfalls werden übermittelte Topics nicht im status angezeigt.
Vermutlich bin ich aber einfach bzgl. der Topics zu dusselig. Außer den selbst gesendeten Topics sehe ich nix im MQTT-Explorer

TheRealMoeder, 2025-02-28:
Hast du neben dem richtigen Topic auch die richtigen JSON Pfade angegeben?

TheRealMoeder, 2025-02-28:
Er sendet keine Daten, er empfängt auf dem genannten Topic nur die Daten, die er als Shelly emulierten soll

Higgy72, 2025-02-28:
Mosquito-broker log:

2025-02-28 20:30:13: New connection from 192.168.0.158:54547 on port 1883.
error: received null username or password for unpwd check
2025-02-28 20:30:13: Client shellypro3em-e8db84dcccf3 disconnected, not authorised.

TheRealMoeder, 2025-02-28:
can you check serial monitor if you user/pass combination shows up in the saved preferences during startup?

TheRealMoeder, 2025-02-28:
[firmware.bin.zip](https://github.com/user-attachments/files/19032684/firmware.bin.zip)

Some issue with the string handling. This should hopefully fix things.

Higgy72, 2025-03-02:
serial at boot shows empty user entry. Password is correct

TheRealMoeder, 2025-03-02:
That is unusual... I will try to look into it

HomeAutoUser, 2025-03-07:
> 0.4.4 (esp8266) funtioniert so weit; Für den ESP32 öffnet sich nach dem Flashen aber der WiFiManager nicht (also kein AP). Habe eben drei verschiedene ESP32 getestet.

Leider kann ich das nicht bestätigen. Habe einen ESP32 geflasht und der WiFI AP öffnete sich.

martind40, 2025-03-07:
@HomeAutoUser welches Dev. Board hast du genutzt?

TheRealMoeder, 2025-03-07:
habe auch nochmal gründlich MQTT user/pass getestet, geht mit 0.4.5 an test.mosquitto.org einwandfrei (ESP8266).

Higgy72, 2025-03-08:
@TheRealMoeder:
Ich danke Dir für Deine Arbeit. 
Leider funktioniert die Nutzer/Passwortanmeldung bei meinem mosquitto unter Homeassistant weiterhin nicht.
Da es bei anderen Nutzern ja zu funktionieren scheint, liegt es wohl an meinem Setup (D1mini, 9 Zeichen User, 20 Buchstaben und Zahlen im Passwort?). Bisher hatte ich damit noch keine Auffälligkeiten.
Ich denke daher, wir können hier zu machen.
Vielleicht finde ich an anderer Stelle eine Lösung meinen Hichi mit dem MS-A2 zu verklöppeln.
Besten Dank!

TheRealMoeder, 2025-03-08:
Das ist die Lösung, User und Passwort sind im Moment (willkürlich) auf 19 Zeichen begrenzt. Erhöhe ich beim nächsten Release auf das doppelte, dann sollte es gehen.

TheRealMoeder, 2025-03-08:
[firmware.bin.zip](https://github.com/user-attachments/files/19145221/firmware.bin.zip)

Hier das 0.4.5 release mit 40 Zeichen für User und Passwort für den ESP8266

HomeAutoUser, 2025-03-10:
@martind40 
ich habe das DEV Board `esp32doit-devkit-v1` genommen um es auf einem AZ Delivery ESP32-WROOM-32 zu compilieren. Ich habe das Projekt über Platform IO selbst compiliert und hochgeladen. -> funktionierte

martind40, 2025-03-10:
@HomeAutoUser , eigenartig; hab sogar dasselbe Board unter anderem verwendet. Hochgeladen hatte ich es einerseits mit dem FlashTool (verschiedene Boutraten) von Espressif und auch mit ESPhome. Die Version für den 8266 Chip funktioniert dagegen pefekt, was das Öffnen des AP betrifft (hier nutze ich ebenso das Flashtool und auch mit der max. Bautrate).

Higgy72, 2025-03-10:
@TheRealMoeder:
Besten Dank! Das war die Ursache. Funktioniert nun problemlos!

TheRealMoeder, 2025-03-10:
Dann mache ich hier erstmal zu, bezüglich der Flash-Probleme bitte am besten einen neuen issue starten.

closed 2025-03-10 by TheRealMoeder

HomeAutoUser, 2025-03-10:
@martind40 

> [@HomeAutoUser](https://github.com/HomeAutoUser) , eigenartig; hab sogar dasselbe Board unter anderem verwendet. Hochgeladen hatte ich es einerseits mit dem FlashTool (verschiedene Boutraten) von [...]

Ich empfehle dir, nutze zum compilieren und hochladen die Software Visual Studio Code / PlatformIO. Mit dem FlashTool habe ich keine guten Erfahrungen gemacht bis dahin gehend, Funktionseinschränkungen / Störungen.