TheRealMoeder/Energy2Shelly_ESP #19: MQTT Daten kommen korrekt am Hoymiles Speicher an - Output fährt auf 800 Watt

fke222, 2025-03-06:
Hallo,
die Integration in die App hat gut funktioniert.
Ich sende alle 2 Minuten neue "Powerwerte" mittel MQTT an den Shelly-Emulator. Dort kommen die Werte an und stelllen den Output des Speichers in der App auf den übertragenen MQTT Wert. Tatsächlich messe ich aber am Ausgang des Speichers 800 Watt.
Dies ist mein Datenaufruf: 
mosquitto_pub -h MQTTBroker -t 'tele/meter/SENSOR' -m  "{\"Energy\":{\"Power\": $esp_power_set,\"In\":0,\"Out\":0}"
Für In und Out setze ich keine Werte (also jeweils 0)  - ist dies so korrekt?
Oder ist mein Speicher defekt?

TheRealMoeder, 2025-03-06:
Wie sieht die Konfiguration in energy2shelly aus? Also jsonpath für Power?

fke222, 2025-03-06:
Ich habe die Pfade im Sketch fest vorgegeben:
`//define your default values here, if there are different values in config.json, they are overwritten.
char input_type[40] = "MQTT";
char mqtt_server[80]="MQTTBrokerIP";
char mqtt_port[6] = "1883";
char mqtt_topic[60] = "tele/meter/SENSOR";
char power_path[60] = "Energy.Power";
char energy_in_path[60] = "Energy.In";
char energy_out_path[60] = "Energy.Out";
char shelly_mac[13] = "B827EB364232";
char shelly_name[26] = "shellypro3em-B827EB364232";
char query_period[10] = "1000";
`
wie schon geschrieben: in der App werden die Werte geändert und als Abgabe vom Speicher dargestellt. Meine Messsteckdose am Ausgang zeigt aber konstant 800 Watt (voreingestellter max.-Wert für Hauseinspeisung).
Die MQTT Werte werden alle 2 Minuten aktualisiert.

[image]

TheRealMoeder, 2025-03-06:
Dann liegt es wahrscheinlich an der langsamen Aktualisierung von 2 Minuten. Der Shelly wird von Speicher mehrmals pro Sekunde abgefragt und rergelt entsprechend nach, da sich aber der Messpunkt ja nicht geändert hat schiebt er andauernd mehr Strom raus. Versuch mal das mqtt topic alle paar Sekunden zu aktualisieren.

fke222, 2025-03-06:
Um die Daten schneller zu senden möchte ich einen ESP32 einsetzen - jetzt habe ich ein Shellscript - das kann mit cronjob nur im Minutentakt (frage mittels SQL DB ab).
Leider stehe ich mit JSON ein wenig auf Kriegsfuss. Wie müsste den der Aufruf von mqttclient.publish aussehen.
Brauche ich ein JSON Objekt oder kann ich das Topic direkt übergeben (wie sieht es dann aus).
Wenn JSON wie soll dies aussehen?

 char Energy2Shelly[250];
 StaticJsonDocument<250> doc;

  doc["Power"] = Power; //Variable Power enthält den neu zu setzenden Powerwert als int
  doc["In"] = "0";
  doc["Out"] = "0";
  serializeJson(doc, Energy2Shelly);
  mqttclient.publish("tele/meter/SENSOR", Energy2Shelly);

Dies funktioniert nicht.
Kannst du bitte helfen. Danke

TheRealMoeder, 2025-03-07:
Sieht alles gut aus, hast du auch vorher mqttclient richtig intialisiert und connected?

fke222, 2025-03-07:
Auch mit 1 Sekunden Intervall (MQTT-Wert an ShellyEmulator) keine Veränderung.
Im Emulator wird der Wert richtig angezeigt.
In der APP auch so.
Tatsächliche Einspeisung aus dem Speicher ist 800 W.

fke222, 2025-03-07:
Beispiel:
mittels ESP32 sende ich für Power 250 Watt.
Im Emulator wird für Power 250 W angezeigt.
Die App zeigt: 
   450 W von der PV, 
   250 W aus dem Speicher ins Netz, (dies sollte es auch sein)
   200 W in den Speicher.
Am Anschluss ins Hausnetz (Schaltsteckdose mit Strommessung) sind 800 W.
Zusätzlich habe ich in dem Stromkreis einen Shelly PM mini - Dieser zeigt ca. 785 W (es gibt einige Verbraucher in dem Stromkreis bis zum Shelly PM mini.
Des weiteren ist in diesem Stromkreis ein Zähler mit S0 Ausgang installiert - der zeigt auch einen ähnlichen Wert an (kann aber nicht zwischen Einspeisung und Verbrauch unterscheiden). 
Funktioniert dies bei dir?

fke222, 2025-03-07:
Auch mein Stromzähler sieht das so.
[image]

TheRealMoeder, 2025-03-07:
und du hast in der app für den einspeisepunkt 0 W als ziel vorgegegeben? Kann nicht ganz nachvollziehen warum das passiert,, bei mir geht es. Zumal die App ja offensichtlich die richtigen daten anzeigt.

Higgy72, 2025-03-08:
offtopic @fke222:
Nutzt Du user/Passwort mit Deinem Mosquitto? Ich habe da Probleme mit der Anmeldung.

fke222, 2025-03-08:
Nutze MQTT ohne Passwort und User

fke222, 2025-03-08:
Ich abe auch noch einen "richtigen" Shelly Pro 3EM - werde in den nächsten Tagen mal einen Testaufbau damit machen - vielleicht hat mein Speicher eine Macke. Mit dem Emulator spare ich mir den Einbau in den Zählerschrank.

fke222, 2025-03-10:
Wollte nur fragen - funktioniert es bei auch nicht? Weil als Bug markiert.

fke222, 2025-03-10:
Habe heute mit meinem Shelly mal Tests gemacht. Resultat war gleich.
Aus meiner Sicht -> Softwarefehler im Speicher. Bei der "alten Firmware" konnte man die "Einspeisung ins Stromnetz" einstellen -> wenn man "0" wählt funktioniert es. Bei meiner akt. installierten SW Version habe ch diesen Menüpunkt gar nicht (gefunden) sondern nur den Einstelliung Ausgangsleistung.
Wenn ich diesen Wert auf 100 stelle (0 geht gar nicht) dann werden immer nur max 100 W aus dem Speicher / PV ins Hausnetz gespeist. Stelle ich den Wert auf ddie zulässigen 800 W dann werden immer 800 Watt eingespeist egal welchn Wert der Shelly vorgibt. Diese Einstellung ist als MIndestlast gedacht - für Hauslast + Netzeinspeisung - so sehe ich es jedenfalls.

[image]
[image]

fke222, 2025-03-10:
In meiner App nicht

[image]

TheRealMoeder, 2025-03-10:
Komisch. Und der Shelly ist dem Hoymiles zugeordnet?

fke222, 2025-03-10:
Ja, die Daten kommen auch in der App an

TheRealMoeder, 2025-03-10:
Vielleicht einfach mal Shelly und hoymiles entfernen aus der App und beides wieder neu hinzufügen

Higgy72, 2025-03-10:
Probiere es mal unter 'Geräte' ganz unten. Dort auf den Shelly klicken. Dann unter dem Bild des Shelly gibt es auch den Eintrag 'Einspeisung ins Stromnetz'. (S-Miles V1.5.0)

fke222, 2025-03-11:
Werde weiter testen - brauche aber auch ein wenig Sonne :-)

TheRealMoeder, 2025-03-12:
Da ja die Werte des ESP korrekt ankommen würde ich hier erstmal das issue schließen.

closed 2025-03-12 by TheRealMoeder