TheRealMoeder/Energy2Shelly_ESP #20: Fragestellung Projekt | Überarbeitung readme

HomeAutoUser, 2025-03-07:
Hallo,
ich versuche mich gerade in das Projekt einzuarbeiten.
Für mich stellt sich die Frage, benötige ich nur einen ESP mit Energy2Shelly_ESP um die Funktionalität mit einem Hoymiles MS-A2 zu gewährleisten?
Muss ich bei der Firmware noch Inputdaten einstellen oder was ist Standardmäßig eingestellt?

Ich würde gern in dem Zusammenhang die Readme anpassen wollen und einen PR generieren um für jedermann es verständlich zu halten.
Welches Format müssen oder dürfen die Inputdaten besitzen?

```
On the captive portal you can currently enter:

   - MQTT Server IP, port and topic; power values on the MQTT topic can be either a raw number or at a configurable JSON field using a JSON Path-style syntax, e.g. "energy.data" for {"energy":{"data":mypowervalue}}.
   - "SMA" to use SMA Energy Meter or Home Manager multicast data
   - "SHRDZM" to use SHRDZM smart meter interface UDP unicast data; please enable UDP broadcasts to the IP of the ESP and port 9522 within SHRDZM
   - "HTTP" to use a generic HTTP input; enter a query URL in the second parameter field which delivers JSON data and define at least the JSON Path for total power
```

Diese Erklärung ist für mich etwas Lückenbehaftet / unvollständig.

LG und vielen Dank für euer Verständnis

TheRealMoeder, 2025-03-07:
Hallo,

der ESP genügt prinzipiell alleine, er übernimmt mDNS broadcast und die eigentliche Arbeit des uni-meter. Dokumentation hatte für mich bislang keine hohe Priorität, da es noch sehr früh in der Entwicklung ist und einiges im Fluss ist. Derzeit arbeite ich u.a. am Dreiphasen Input, sodass sich auch die Konfigurationsseite des WiFi Manager noch stetig ändert. Wenn du aber Lust hast. Zur Dokumentation beizusteuern, würde ich das sehr begrüßen. Insbesondere ein paar Bilder würden sicherlich helfen.

HomeAutoUser, 2025-03-10:
Hallo @TheRealMoeder,
gern bin ich behilflich die Doku zu überarbeiten um es für jedermann verständlich zu halten.
Um nicht durcheinander mit Bugs oder Erweiterungen zu kommen, könntest du das Label von dem Issues abändern für das Verständnis? Alternativ würde ich es auch selbst machen, sollte man mir die Rechte erweitern vom Projekt. So kenne ich es von anderen Projekten wie RFHEM oder SIGNALduino.

Wie Vorgehensweise würde ich als PR Vorschlagen, wo du drüber schauen kannst und wo wir ggf. über Formulierungen oder andere Dinge diskutieren könnten bevor man es übernimmt in das Projekt.

HomeAutoUser, 2025-03-10:
@TheRealMoeder hier wäre schon mal ein Vorschlag https://github.com/HomeAutoUser/Energy2Shelly_ESP/tree/main im ersten Schritt

Thx für die Labelkennzeichnung.

TheRealMoeder, 2025-03-10:
Das sieht doch schon mal sehr gut aus, ein paar Kleinigkeiten würde ich noch anpassen, aber ein großer Schritt vorwärts. Pull request kannst du gerne machen sobald du "fertig" bist, kann ich dann zeitnah mergen.

TheRealMoeder, 2025-03-10:
@HomeAutoUser Funktioniert der ESP32 problemlos mit dem Hoymiles Speicher? Dann kann ich #8 zu machen

HomeAutoUser, 2025-03-10:
@TheRealMoeder das kann ich erst gegen Ende der Woche berichten. Die Funktion teste ich und gebe eine Rückmeldung.

Folgende Punkte fielen mir bisher aus
- ESP32 …./reset Funktion geht nicht, er geht nicht in den AP- Mode
- volle Kompatibilität zum Shelly Pro 3EM ist noch nicht gewährleistet (eine angelegtes ShellyDevice in FHEM kann manche Befehle ich abfragen) -> versuche vollständig dahinter zu kommen und sammle Fakten

TheRealMoeder, 2025-03-10:
AP mode am ESP32 muss ich nach schauen.

Volle Kompatibilität war nie das Ziel, da die API Recht umfassend ist. Prinzipiell möglich, aber wie uni-meter wollte ich nur eine schnelle lösung für den Hoymiles Speicher umsetzen.

TheRealMoeder, 2025-03-10:
#24 ausgegliedert.

HomeAutoUser, 2025-03-11:
@TheRealMoeder 

Ich würde für die Inputs noch Beispiele ergänzen wollen.
- "MQTT" (MQTT -JSON String)
- "HTTP" (HTTP String)
- "SMA" for SMA EM/HM Multicast ?
- "SHRDZM" for SHRDZM UDP data ?

Hast du in deinen Aufzeichnungen da Mitschriften? Ich wollte nun das Input via MQTT testen aber erhalte noch kein Ergebnis.

TheRealMoeder, 2025-03-11:
Im  Falle von mqtt gehört da die IP des Brokers rein. SMA, HTTP und SHRDZM ist nur für diese Betriebsarten einzutragen. #18 ist noch auf meiner to do Liste um die Konfiguration eindeutiger hinzubekommen.

HomeAutoUser, 2025-03-11:
@TheRealMoeder so machen wir es.
Ich hoffe, das wir uns nicht missverstehen, da es sich um genau 2 verschiedene Punkte in der Dokumentation handelt.

Hier https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/20#issuecomment-2713976450 (oben) meinte ich das Input, damit der Energy2Shelly_ESP berechnen kann und ausgibt bzw. weitergibt und zum anderen gibt es die Einstellungen im WifiManager. Beides baut auf sich auf für eine richtige Verarbeitung. Diese Frage kam direkt auf, da ich versuchte via MQTT die Daten zu senden. Leider gelang mir kein plausibles Input in Form von Wert oder JSON-String, das man das Ergebnis (nicht 0) im Fenster `/status` entnehmen kann.

HomeAutoUser, 2025-03-19:
@TheRealMoeder, Hallo :-)
[Wie sicher bist du mit diesen Einträgen?](https://github.com/TheRealMoeder/Energy2Shelly_ESP/blob/471ae3410c95d461cbd729f78f5854d9a33c3be3/src/main.cpp#L194:L204)
Bei meinen Ergebnissen, bin ich zu der Erkenntnis gekommen, das `"result"` nicht richtig sein kann.
Sobald ich diesen Wert überall entferne erhalte ich plausible Auswertungen der DeviceInfo. Woher hast du diese Verschachtelung übernommen?

Ich würde 
```
  JsonDocument jsonResponse;
  jsonResponse["id"] = rpcId;
  jsonResponse["src"] = shelly_name;
  jsonResponse["result"]["name"] = shelly_name;
  jsonResponse["result"]["id"] = shelly_name;
  jsonResponse["result"]["mac"] = shelly_mac;
  jsonResponse["result"]["slot"] = 1;
  jsonResponse["result"]["model"] = shelly_model;
  jsonResponse["result"]["gen"] = 2;
  jsonResponse["result"]["fw_id"] = shelly_fw_id;
  jsonResponse["result"]["ver"] = "1.4.8";
  jsonResponse["result"]["app"] = shelly_app;
  jsonResponse["result"]["auth_en"] = false;
  jsonResponse["result"]["profile"] = power_variant;
  serializeJson(jsonResponse,serJsonResponse);
  DEBUG_SERIAL.println(serJsonResponse);
```
ändern, da analog bei der ShellyPlugS (Hardware besitze ich) folgendes heraus kommt und auch dann erst, die Auswertung in FHEM richtig funktioniert. ;-)

**Auszug Hardware ShellyPlugS** 
```
{
    "name":null,
    "id":"shellyplusplugs-d48afc564d04",
    "mac":"D48AFC564D04",
    "slot":1,
    "model":"SNPL-00112EU",
    "gen":2,
    "fw_id":"20241011-114442/1.4.4-g6d2a586",
    "ver":"1.4.4",
    "app":"PlusPlugS",
    "auth_en":false,
    "auth_domain":null
}
```

Ergebnis wäre bei mir:
```
{
  JsonDocument jsonResponse;
  jsonResponse["name"] = shelly_name;
  jsonResponse["id"] = rpcId;
  jsonResponse["mac"] = shelly_mac;
  jsonResponse["slot"] = shelly_slot;
  jsonResponse["model"] = shelly_model;
  jsonResponse["gen"] = shelly_gen;
  jsonResponse["fw_id"] = shelly_fw_id;
  jsonResponse["ver"] = shelly_ver;
  jsonResponse["app"] = shelly_app;
  jsonResponse["auth_en"] = false;
  jsonResponse["auth_domain"] = nullptr;
  jsonResponse["profile"] = power_variant;
  serializeJson(jsonResponse,serJsonResponse);
  DEBUG_SERIAL.println(serJsonResponse);
}
```

---> Hinweise sind ebenso
https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Shelly/#http-endpoint-shelly
https://shelly-api-docs.shelly.cloud/gen2/ComponentsAndServices/Shelly/#shellygetdeviceinfo

Zwischenstand FW zum testen für Dich,
https://github.com/HomeAutoUser/Energy2Shelly_ESP/tree/comp_FHEM
da du die bisher falsche Void auch bei webSocketEvent nutzt.

TheRealMoeder, 2025-03-20:
Ich passe mal die http-antworten entsprechend in meinem repository an damit es sauber bleibt. für die anderen kleineren Änderungen wie power variant und von mir aus auch die saubere Auflistung der Voreinstellungen zu beginn wäre ein PR super. dann führen wir das zusammen und können weiter mit FHEM und hoymiles parallel testen.

TheRealMoeder, 2025-03-20:
#26 works with hoymiles storage and also gives correct http responses I hope.

HomeAutoUser, 2025-03-20:
@TheRealMoeder Bitte nicht so viel anpassen.
Diese Basis https://github.com/HomeAutoUser/Energy2Shelly_ESP/tree/comp_FHEM ist der Stand, wo wir vereinen können.
Du warst mit dem #27 etwas voreilig.

Gern kann ich auch deine Anliegen / Anpassungen bei mir vornehmen und dann vollende ich einen PR auf dich.

TheRealMoeder, 2025-03-20:
Die Anpassungen on #26 waren inhaltlich erforderlich, deswegen habe ich die direkt gemacht. Hatte mich schon länger gestört.

#27 war primär gedacht um die Änderungen besser kommentieren zu können. Leider missverstehst du die Verwendung von TRIPHASE in meinem Code - hier ist explizit nicht gedacht zwischen den verschiedenen Profilen des Shelly Pro 3EM zu differenzieren (mono vs triphase), sondern es geht nur darum, ob im triphase Profil die Quelldaten monophasisch ankommen und gedrittelt werden auf die Phasen oder ob mit dreiphasigen Quelldaten gearbeitet werden kann. Deswegen sehe ich keinen Mehrwert darin, das Monophase Profil zu implementieren.

HomeAutoUser, 2025-03-20:
Ich habe den Stand von mir mit deinem main Branch vereint händig und lasse soeben alle Varianten durchlaufen zum testen intern.
Wenn diese ohne Fehler sind, mache ich dir einen PR welcher auf dem Stand von deinem basieren. Darauf können wir aufbauen und haben die Erweiterung HTTP -> FHEM implementiert bis auf eine kleine automatische Anpassung der Werte.

Die Unterscheidung mit TRI und mono gedenke ich zu verstehen was du meinst. Leider ist für die Verarbeitung FHEM dort beides notwendig. Grund Hierfür sind unterschiedliche HTTP Requests mit unterschiedlichen Namen. Wenn man eine Variante weg lässt, so kann es passieren, das das System aufgrund von fehlenden Werte WARNUNGEN bis hin zu ABBRÜCHEN bringt. Daher die vollständige Integrierung mit unterschiedlichen Namen. Mit allen Kommentaren solltest du dann durchschauen denke ich.

Schau mal drüber. 
https://github.com/HomeAutoUser/Energy2Shelly_ESP/tree/main_comp_FHEM

TheRealMoeder, 2025-03-20:
Ich erstelle mal einen PR um die Kommentare nachvollziehbarer zu machen.

Crusher606, 2025-03-30:
Ich habe die HTTP Variante gestern getestet.
Sie funktioniert mit meinem alten Shelly 3EM und meinem Marstek Venus E per Energy2Shelly auf einem ESP 8266.

Wenn du die Anleitung zu diesem Punkt erweitern möchtest, findest du unter „Add UDP output for Marstek Venus #11“ die Informationen zu dem Testverlauf.

Ich habe die *.bin geflasht

Im AP folgende Daten eingegeben:

Bei General Settings:
- - - - - - -
mein WLAN mit Passwort

HTTP

query URL for generic HTTP:
http://IP-DES-SHELLY-3EM/status

Bei JSON Path:
- - - - - - - -
TRIPHASE

Phase 1 Power:
emeters[0].power

Phase 2 Power:
emeters[1].power

Phase 3 Power:
emeters[2].power

uputer, 2025-05-12:
Servus zusammen,

vielleicht passt das jetzt hier nicht unbedingt rein.
Aber: Ich möchte mich auf diesem Weg noch einmal für eure hervorragende Arbeit und Euren Support bedanken.
Der Energy2Shelly läuft bei mir (auf einem 8266, vorher Ahoy-DTU) super gut.
War ich anfangs noch am Überlegen, mir zur Kopplung mit dem Venus E mir einen Shelly EM3 pro zu besorgen, verschwende ich mittlerweile daran keinen Gedanken mehr.

Der Energy2Shelly läuft so gut, das ich keine Alternative benötige.

Super! Danke Euch noch mal!

Gruß