TheRealMoeder/Energy2Shelly_ESP #24: /reset mit working on ESP32

TheRealMoeder, 2025-03-10:


HomeAutoUser, 2025-03-11:
Ich habe heute versucht dem Fehler auf den Sprung zu kommen. Leider habe ich noch keinen richtigen Ansatz.
In manchen Foren liest man, das `ESP.restart()` manchmal nicht funktioniert. In dem Testfalle heute, habe ich nach dem uC löschen diesen eingestellt mit Wifi und nach dem verbinden ein ` /reset` ausgefführt um in den AP-Mode zu gelangen. Einmal gelang es.... Die anderen Male durchläuft er den Programmcode aber bleibt nach einem Neustart im davor eingestellten WiFi.

TheRealMoeder, 2025-03-11:
Ich denke ich man muss einfach auf wifiManager.resetSettings(); umändern dann sollte es gehen, dort wird automatisch nach esp32 und esp8266 differenziert. Hatte da anfangs Probleme mit als ich noch SPIFFS in Einsatz hatte, deshalb flog das damals wieder raus aus dem Code.

TheRealMoeder, 2025-03-18:
please retest with 0.4.6 release

HomeAutoUser, 2025-03-18:
Ich habe es soeben getestet.
Mit dem Befehl `/reset` gelang er wieder in den AP Mode.

PS: Ich reiche Dir demnächst ein paar Anpassungen nach welche hilfreich sein könnten weil ich den HTTP Request zur Komptabilität -> FHEM beleuchte.
`- /shelly` failed und andere ...

TheRealMoeder, 2025-03-18:
Perfekt, ich freue mich über weiteren Input und machte hier erstmal zu.

closed 2025-03-18 by TheRealMoeder