TheRealMoeder/Energy2Shelly_ESP #33: Tasmota Hichi Kopf über HTTP auslesen

uputer, 2025-03-31:
Servus,

super Idee von dir / Euch!

Ich habe das gerade einmal auf meinem alten 8266 installiert. Funktioniert soweit, allerdings wird nur V / Hz ausgelesen.

(Quelle: Hichi Lesekopf)

Die momentane Power (W) bekomme ich nicht ausgelesen.

Die Einträge _Energy. Power_ oder auch nur _Power_ funktionieren als Jason Path nicht.

{
  "id": 1,
  "src": "shellypro3em-441793150258",
  "dst": "user_1",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 230,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0,
    "b_voltage": 230,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0,
    "c_voltage": 230,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
  }
}

Der Hichi hat folgende Einträge:

>D
>B
->sensor53 r
>M 1
+1,3,s,0,9600,Main,1,10
1,77070100010800FF@1000,Total Consumed,kWh,counter_pos,3
1,77070100020800FF@1000,Total Feed,kWh,counter_neg,3
1,77070100100700FF@1,Power,W,power,0
#

Im HA ist folgendes aufgeführt:

**  "Time": "2025-03-31T18:53:26",
  "Time": "2025-03-31T18:53:36",
  "Main": {
    "counter_pos": 4590.616,
    "counter_neg": 821.601,
    "power": 19
    "power": 27**

Kann mir da jemand mal unter die Arme greifen, was ich eintragen muss?

Vielen Dank!

TheRealMoeder, 2025-03-31:
Main.power

uputer, 2025-03-31:
Leider nein. Resultat bleibt gleich.
Wo zieht er sich dann V / Hz?

uputer, 2025-04-01:
in OpenDTUob ziehe ich die Daten mit:

[image]

TheRealMoeder, 2025-04-01:
Die URL für die tasmota Ausgabe ist korrekt. Poste mal das Ergebnis, wenn du die URL in einem Browser öffnest.

uputer, 2025-04-01:
> Die URL für die tasmota Ausgabe ist korrekt. Poste mal das Ergebnis, wenn du die URL in einem Browser öffnest.

{"StatusSNS":{"Time":"2025-04-01T11:00:04","Main":{"counter_pos":4596.574,"counter_neg":822.166,"power":883}}}

TheRealMoeder, 2025-04-01:
Dann lautet der Eintrag für Power Path logischerweise `StatusSNS.Main.power`

Und Energy from grid ist dann `StatusSNS.Main.counter_pos` und Energy to grid muss `StatusSNS.Main.counter_neg` sein.

uputer, 2025-04-01:
wäre so schön gewesen ...

{
  "id": 1,
  "src": "shellypro3em-441793150258",
  "dst": "user_1",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 230,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0,
    "b_voltage": 230,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0,
    "c_voltage": 230,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
  }
}

Statt dem Punkt habe ich es auch mit Komma, Semikolon und Schrägstrich probiert.

no chance :(

TheRealMoeder, 2025-04-01:
Mach mal einen Screenshot von der Konfigurationsseite im Energy2shelly

uputer, 2025-04-02:
[image]

Ergebnis:

{
  "id": 1,
  "src": "shellypro3em-",
  "dst": "user_1",
  "result": {
    "id": 0,
    "a_current": 0,
    "a_voltage": 230,
    "a_act_power": 0,
    "a_aprt_power": 0,
    "a_pf": 1,
    "a_freq": 50,
    "b_current": 0,
    "b_voltage": 230,
    "b_act_power": 0,
    "b_aprt_power": 0,
    "b_pf": 1,
    "b_freq": 50,
    "c_current": 0,
    "c_voltage": 230,
    "c_act_power": 0,
    "c_aprt_power": 0,
    "c_pf": 1,
    "c_freq": 50,
    "total_current": 0,
    "total_act_power": 0,
    "total_aprt_power": 0
  }
}

HTTP Ist ausgewählt

TheRealMoeder, 2025-04-02:
Und der obere Teil der Konfiguration?

TheRealMoeder, 2025-04-02:
Ah. Fehler gefunden. `power`mit kleinem`p`

uputer, 2025-04-02:
> Ah. Fehler gefunden. `power`mit kleinem`p`

Leider ... nein

Tut mir echt leid das ich dich so belästige ....

[image]
[image]

TheRealMoeder, 2025-04-02:
Die Anführungszeichen um `HTTP` müssen weg. Das ist vielleicht etwas missverständlich ausgedrückt. Und bei query period mal 1000 eintragen

uputer, 2025-04-03:
**Works!!!**

Die Anführungszeichen ... das war es 

Der Ausschlag an der Venus sieht jetzt ganz anders aus.

Links der B2500 Emulator, rechts der Energy2Shelly

[image]

uputer, 2025-04-03:
Nachdem der Venus ein wenig rumzickte, habe ich noch ein wenig rumgespielt.

Venus durch drücken des Aus Knopfes (für 60 sec) komplett aus und an geschaltet (Stecker draußen).

In die Konfig Seite des Energy2Shelly noch die Mac des 8266 eingetragen.

Siehe da, nun sieht das deutlich besser aus.

[image]

Der Venus E reagiert deutlich schneller.
Liegt aber auch daran, das die Abfrage über HTTP wesentlich schneller erfolgt wie über die MQTT Abfrage des Hichi Lesekopfes am Zähler (min. alle 10sec) Plus wohl der Umweg über HA, MQTT & den B2500 Emulator.

Good job Ihr Coder!!!

Somit sollte es über den Hichi Lesekopf so aussehen (MAC muss noch eingetragen werden):

[image]

[image]

TheRealMoeder, 2025-04-03:
Die Mac ist normalerweise vorausgefüllt, ging vielleicht irgendwann bei den Konfigurationsversuchen verloren. Freut mich das es klappt und vielen Dank für das Feedback, vielleicht können wir das ja in die Doku hier mit aufnehmen.

Zum Thema dreiphasig vs einphasig: du hast nur die Summenleistung als Quelldaten aus dem Hichi, deswegen verteilt Energy2shelly die gleichmäßig auf alle drei Phasen. Das setzt natürlich voraus, dass der Marstek nicht nur für eine Phase konfiguriert ist sondern auch auf die Summer der drei Phasen steuert.

closed 2025-04-03 by TheRealMoeder