TheRealMoeder/Energy2Shelly_ESP #34: Kompatibilität für mehr als einen Speicher

andy5macht, 2025-04-01:
Hallo zusammen.
Danke für dieses Projekt! Das hat mal richtig Potential!

Was für mich noch wichtig währe:
Wenn man zwei Marstek Venus hat, dann kommt man mit einem Shelly nicht hin beide parallel laufen zu lassen.
Hier ist meine Überlegung eine Variable mit der Anzahl der Speicher einzugeben.
Bei z.B. zwei Speichern mit gleich großer Kapazität könnte man die Ausgabe der Leistung jeweils durch 2 teilen.
So würden die Speicher parallel betrieben werden können.

Würde mich freuen wenn das umsetzbar wäre!

TheRealMoeder, 2025-04-01:
Hallo und vielen Dank. Machbar ist das prinzipiell, allerdings hat Sdeigm so etwas beim Uni-Meter schon probiert und es kam zu Oszillationen sodass irgendwann ein Speicher den anderen lädt. Ohne Rückkanal zur Speicherregelung geht das also nicht sinnvoll.

andy5macht, 2025-04-01:
Ist es aktuell möglich einen Shelly mit "Energy2Shelly_ESP" auszulesen? Ich denke das dürfte sicherlich gehen.
Hier Wäre meine Idee mit zwei Marstek Venus:
- 1. Venus verwendet eine Phase direkt vom Shelly Pro 3EM
- 2. Venus wird über dieses Projekt mit den anderen beiden Phasen "versorgt"

Wäre das eine Idee, zwei dieser Speicher zu bedienen?
- Ist es mit diesem Projekt möglich zwei Phasen zu lesen und als 3 Phasen auszugeben (eine Wäre halt immer 0)
- Oder zwei Phasen zu einer Phase aufzusummieren?

TheRealMoeder, 2025-04-01:
Man könnte einem esp als Quelldaten eine Phase vom 3EM geben und der splittet diese auf 3 Phasen auf. Dann einen zweiten ESP der das mit einer weiteren Phase macht. Wäre prinzipiell auch auf einem esp möglich, aber etwas aufwändiger umzusetzen. Vielleicht erstmal mit zwei esps testen. Die Konfiguration wäre dann:

Http query url: `http://ipdesshelly3e/status`
power path: `emeters[0].power` (für L1, bzw. 1 und 2 für die anderen beiden Phasen)

Allerdings musst du hierfür entweder den testbuild aus #11 vom 24.3. nehmen, selber aus Main kompilieren oder auf Release 0.5 warten.

andy5macht, 2025-04-01:
Ja super, vielen Dank!
Der zweite Marstek kommt erst noch. Von demher habe ich noch keine Eile.

closed 2025-04-01 by TheRealMoeder