TheRealMoeder/Energy2Shelly_ESP #35: Modbus TCP für Kostal Smart Meter (KSEM)

andy5macht, 2025-04-01:
Fronius uns SMA sind hier schon unterstützt. Da die Wechselrichter eine API besitzen, mit der man recht einfach die Leistung abfragen kann.
Wäre es möglich Modbus TCP (SunSpec) mit einzubinden? Hiermit wären eine Vielzahl an Wechelrichter bzw. Smartmeter ansprechbar.

TheRealMoeder, 2025-04-01:
Technisch sicher kein Problem, entsprechende Libraries für ESP existieren. Ich habe allerdings persönlich im Moment keine Zeit, das umzusetzen.

GalthrikHit, 2025-04-13:
Hello, I've implemented a basic sunspec solution. Works fine with my KSEM
But I am new to this git stuff. So just past here my changes. Okay just a fast hack.
Entering SUNSPEC in the config (first row) and IP adress of the KSEM in the MQTT adress, will do the job.
------------------------------------------------------

#include <ModbusIP_ESP8266.h>

IPAddress SUNSPEC_IP;
ModbusIP mb1;

bool dataSUNSPEC = false;

In setup()

  if (dataSUNSPEC) {
    mb1.client();
    SUNSPEC_IP.fromString(mqtt_server);
    if (!mb1.isConnected(SUNSPEC_IP)) {  // reuse mqtt server adresss for modbus adress
      mb1.connect(SUNSPEC_IP, 502);
      Serial.println("Try to connect powermeter");
    }
  }

also in WifiManagerSetup
...
} else if (strcmp(input_type, "SUNSPEC") == 0) {
    dataSUNSPEC=true;
    DEBUG_SERIAL.println("Enabling generic SUNSPEC data input");
  }
....
sint16_t res1[256];
void parseSUNSPEC() {
  #define SUNSPEC_BASE 40071
  SUNSPEC_IP.fromString(mqtt_server);
  if (!mb1.isConnected(SUNSPEC_IP)) {
    mb1.connect(SUNSPEC_IP, 502);
  } else {
    uint16_t transaction = mb1.readHreg(SUNSPEC_IP, SUNSPEC_BASE, (uint16_t*) &res1[0], 64, nullptr, 71);
    delay(100);
    mb1.task();
    int t = 0;
    while (mb1.isTransaction(transaction)) {
      mb1.task();
      delay(10);
      t++;
      if (t > 500) {
        DEBUG_SERIAL.println("Timeout SUNSPEC");
        mb1.disconnect(SUNSPEC_IP);
        break;
      }
    }
    sint32 power=0;
    if (t<=500) {
      for (int n=0;n<3;n++) {
        PhasePower[n].power=res1[40088-SUNSPEC_BASE+n];
        PhasePower[n].apparentPower=res1[40092-SUNSPEC_BASE+n];
        PhasePower[n].current=res1[40093-SUNSPEC_BASE+n]*0.01;
        PhasePower[n].powerFactor=res1[40103-SUNSPEC_BASE+n]*0.001;
        PhasePower[n].voltage=res1[40077-SUNSPEC_BASE+n]*0.01;
        PhasePower[n].frequency=res1[40085-SUNSPEC_BASE]*0.01f;
        power+= PhasePower[n].power;
      }

    }
    DEBUG_SERIAL.printf("SUNSPEC power: %d,%d\n\r", t,power);
  }
}

in main loop, reuse time period from http (1s per default)

 if (dataSUNSPEC) {
     currentMillis = millis();
    if (currentMillis - startMillis >= period) {
       parseSUNSPEC();
      startMillis = currentMillis;
    }
   
  }

TheRealMoeder, 2025-04-14:
Very nice, thanks! Did you suffer from any timeouts? Not sure what the effect of blocking the esp for up to 500 ms could have, as the Hoymiles storage requests power every 500 ms

GalthrikHit, 2025-04-14:
Hello,

this is a reuse of code. KSEM is not really an issue. Pretty fast. But SMA MODBUS is not working well on my side. So needed some extra timeout handling. You could easily set the timeout counter "t" to max 10 or so. In my debug out, I've never seen t>=1.
In my project I've used "task"/"leantask" scheduler. So every "delay" function-call will yield into an handover to the other tasks. It is cooperative scheduler. Also some yield() here and there, keeps the system responsive.
=> Many tasks in parallel

Maybe you can add two other parameter for modbus into the main config. Since I've set the modbus TCPIP port to 502 and the modbus device number to 71 in mb1.readHreg(SUNSPEC_IP, SUNSPEC_BASE, (uint16_t*) &res1[0], 64, nullptr, 71);

Both parameters should be configurable.

The scaling factors are fixed to 10.0, 0.01, 0.001 and so on. The sunspec registers defines the scaling values. Could be dynamic. But I found no SUNSPEC implementation with other than these fixed scaling values.

I've have a project with esp32, here the #include <ModbusIP_ESP8266.h> works fine, too.

TheRealMoeder, 2025-04-14:
If you could test everything I would integrate this into the master branch

andy5macht, 2025-04-14:
> The scaling factors are fixed to 10.0, 0.01, 0.001 and so on. The sunspec registers defines the scaling values. Could be dynamic. But I found no SUNSPEC implementation with other than these fixed scaling values.

Fronius does use dynamic scaling values.

GalthrikHit, 2025-04-14:
Ok,

please have a look on the attached file.
Current power and so on, is ok.
Summed energy is strange. could not figure out why. Consumption is ok, but not feed in.

[main.txt](https://github.com/user-attachments/files/19736011/main.txt)

TheRealMoeder, 2025-05-05:
Implemented, tested and merged. Thank again @GalthrikHit

closed 2025-05-05 by TheRealMoeder