TheRealMoeder/Energy2Shelly_ESP #41: Request for flash positions for ESP32

daniel62390, 2025-05-01:
First of all, many thanks for the great idea to port the uni-meter to the ESP. I would like to try the tool with the Growatt Noah 2000 and an ESP32. When flashing the .bin-files from the last release with the Espressif Download Tool, I was asked for positions for the partition, bootloader and firmware files. Standard positions (e.g. 0x1000) did not work. Could you maybe provide the flash positions? Many thanks! -Daniel

TheRealMoeder, 2025-05-04:
I use this partition scheme for the release binary:
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv

TheRealMoeder, 2025-05-10:
0x1000 bootloader.bin
0x8000 partitions.bin
0xe000 boot_app0.bin
0x10000 firmware.bin

[boot_app0.bin.zip](https://github.com/user-attachments/files/20138641/boot_app0.bin.zip)

daniel62390, 2025-05-10:
Thank you, flashing an ESP32 with these parameters using the Espressif Download Tool was succesful. Configuration also went well, but the emulated Shelly is not found by the Growatt Noah 2000 (is also not listed on the main page). If in the future a tester is needed, feel free to contact me.

TheRealMoeder, 2025-05-10:
Can you post serial debugging output from the esp when trying to find it with the Noah2000?

daniel62390, 2025-05-11:
When scanning the network with the ShinePhone App from the Growatt Noah (and setting the checkbox that Noah and Shelly are in the same local network), there is nothing special visible in the serial output, neither for a scan for a Shelly 3EM nor a Shelly Pro 3EM:
12:12:56.723 -> Querying HTTP source
12:12:56.977 -> Resolving monophase (single path only)
12:12:56.977 -> Current total power: 1.29
12:13:01.721 -> Querying HTTP source
12:13:02.022 -> Resolving monophase (single path only)
12:13:02.022 -> Current total power: 0.14
12:13:06.737 -> Querying HTTP source
12:13:06.984 -> Resolving monophase (single path only)
12:13:06.984 -> Current total power: -0.47
12:13:11.726 -> Querying HTTP source
12:13:12.228 -> Resolving monophase (single path only)
12:13:12.228 -> Current total power: -3.20
12:13:16.746 -> Querying HTTP source
12:13:17.000 -> Resolving monophase (single path only)
12:13:17.000 -> Current total power: 0.84
12:13:21.736 -> Querying HTTP source
12:13:21.999 -> Resolving monophase (single path only)
12:13:21.999 -> Current total power: -2.15
12:13:26.715 -> Querying HTTP source
12:13:27.016 -> Resolving monophase (single path only)
12:13:27.016 -> Current total power: 3.32
12:13:31.731 -> Querying HTTP source
12:13:31.978 -> Resolving monophase (single path only)
12:13:31.978 -> Current total power: -2.60
12:13:36.754 -> Querying HTTP source
12:13:37.001 -> Resolving monophase (single path only)
12:13:37.001 -> Current total power: -4.81
12:13:41.734 -> Querying HTTP source
12:13:42.019 -> Resolving monophase (single path only)
12:13:42.019 -> Current total power: 0.03
12:13:46.755 -> Querying HTTP source
12:13:47.010 -> Resolving monophase (single path only)
12:13:47.010 -> Current total power: -1.12
12:13:51.748 -> Querying HTTP source
12:13:51.985 -> Resolving monophase (single path only)
12:13:51.985 -> Current total power: 5.59
12:13:56.753 -> Querying HTTP source
12:13:56.989 -> Resolving monophase (single path only)
12:13:56.989 -> Current total power: -2.13
12:14:01.711 -> Querying HTTP source
12:14:02.560 -> Resolving monophase (single path only)
12:14:02.560 -> Current total power: -4.80
12:14:06.711 -> Querying HTTP source
12:14:07.012 -> Resolving monophase (single path only)
12:14:07.012 -> Current total power: -1.13
12:14:11.719 -> Querying HTTP source
12:14:12.020 -> Resolving monophase (single path only)
12:14:12.020 -> Current total power: -0.19
12:14:16.741 -> Querying HTTP source
12:14:17.259 -> Resolving monophase (single path only)
12:14:17.259 -> Current total power: -36.46
12:14:21.712 -> Querying HTTP source
12:14:22.004 -> Resolving monophase (single path only)
12:14:22.004 -> Current total power: -47.16

The total power is around 0W because there is OpendtuOnBattery running and limiting the Hoymiles inverter. I set the HTTP query time to 5s.

TheRealMoeder, 2025-05-11:
That is odd. This is the first time I've heard of an mDNS broadcast issue. Do you by chance have an esp8266 to test with? mDNS code is slightly different on this platform. 
Or are you possibly using a guest wifi network where clients are isolated from each other?

daniel62390, 2025-05-12:
I have an ESP8266. How do I have to flash the .bin file here?
All devices have been in the same wifi network (no guest network) from a fritzbox (Fritz OS 8). Clients are not isolated. The HTTP communication with the Tasmota device on the power meter was working, same as between OpenDTUonbattery and the Tasmota.

TheRealMoeder, 2025-05-14:
> I have an ESP8266. How do I have to flash the .bin file here?
Nothing special for the esp8266, just use your favorite flash tool and the provided release bin.

daniel62390, 2025-05-15:
Thank you, flashing the ESP8266 .bin-file with the adress 0x00000 worked.

As before, I see no hint of the Growatt Noah in the serial output:

19:43:38.879 -> Querying HTTP source
19:43:39.133 -> Resolving monophase (single path only)
19:43:39.133 -> Current total power: 6.03
19:43:39.882 -> Querying HTTP source
19:43:40.151 -> Resolving monophase (single path only)
19:43:40.151 -> Current total power: 3.47
19:43:40.873 -> Querying HTTP source
19:43:41.157 -> Resolving monophase (single path only)
19:43:41.157 -> Current total power: -1.18
19:43:41.859 -> Querying HTTP source
19:43:42.161 -> Resolving monophase (single path only)
19:43:42.161 -> Current total power: -0.70
19:43:42.868 -> Querying HTTP source
19:43:43.165 -> Resolving monophase (single path only)
19:43:43.165 -> Current total power: 4.39
19:43:43.868 -> Querying HTTP source
19:43:44.122 -> Resolving monophase (single path only)
19:43:44.122 -> Current total power: 1.38

Under <ip-address>/status, the first line says "id | 0", although I did not change the default Shelly ID in the WIFI config menu. Is this correct? "total_act_power | -9.21" looks good, as well as the values for the 3 phases, which seem to be 1/3 of the total power each.

Is there someone else with a Growatt Noah 2000, who can try this as well?

TheRealMoeder, 2025-05-15:
Id 0 in the context of /status is just fine. If you go to /rpc/Shelly.GetDeviceInfo you should correctly see your id.

TheRealMoeder, 2025-05-15:
I tried to read up on this issue, it seems the Noah is not directly querying the Shelly on the local network but using the Shelly Cloud?

daniel62390, 2025-05-16:
Yes, unter /rpc/Shelly.GetDeviceInfo I see the correct shelly ID :)

You are right. In the past, the Noah App asked for login data for the Shelly Cloud for connecting a shelly. But some weeks ago, they changed that. Now, for connecting a Shelly, one has to set a tick "Your device and Shelly Pro 3EM are already under the same local network" and confirm with OK. Then the search is started. There is no login data required anymore. Therefore I was hoping that energy2shelly also works for the Noah. Maybe I find some time to dig into mDNS and how the Noah searches for a shelly.

daniel62390, 2025-05-17:
It seems that the Shinephone app for the Growatt Noah is still buggy regarding the local discovery of shelly devices: https://github.com/robertzaage/GroBro/issues/43

TheRealMoeder, 2025-05-17:
Thanks for researching. mDNS is working fine for others like Hoymiles, ioBroker and even Openhab (although other parts of Shelly API aren't implemented yet so it is not fully working). Let's wait and see what future Growatt updates have to offer on this issue.

daniel62390, 2025-07-06:
With the latest firmware version of the Noah 2000 (V13.10.XX.4015), the emulated shelly (ESP32 used) is found by the Growatt app.

11:48:09.122 -> Querying HTTP source
11:48:09.424 -> Resolving monophase (single path only)
11:48:09.424 -> Current total power: -13.69
11:48:09.424 -> Total consumption: 13849.62 - Total Grid Feed-In: 717.87
11:48:12.463 -> {"name":"shellypro3em-240ac4829db0","id":"shellypro3em-240ac4829db0","mac":"240ac4829db0","slot":1,"model":"SPEM-003CEBEU","gen":"2","fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"profile":"triphase"}
11:48:14.126 -> Querying HTTP source
11:48:14.626 -> Resolving monophase (single path only)
11:48:14.626 -> Current total power: -5.01
11:48:14.626 -> Total consumption: 13849.62 - Total Grid Feed-In: 717.87
11:48:15.377 -> {"name":"shellypro3em-240ac4829db0","id":"shellypro3em-240ac4829db0","mac":"240ac4829db0","slot":1,"model":"SPEM-003CEBEU","gen":"2","fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"profile":"triphase"}
11:48:18.342 -> {"name":"shellypro3em-240ac4829db0","id":"shellypro3em-240ac4829db0","mac":"240ac4829db0","slot":1,"model":"SPEM-003CEBEU","gen":"2","fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"profile":"triphase"}

However, the app shows just 0W, 0kWh, 0A, 0V from the shelly. I tried already both UDP-ports in the config. Does anyone has an idea?

daniel62390, 2025-07-08:
I found this in the uni-meter github: https://github.com/sdeigm/uni-meter/issues/160
Looks like the issue is known and already solved in the latest uni-meter version. In case you plan to update the energy2shelly, I would be happy to test it and give feedback.

TheRealMoeder, 2025-07-09:
I saw that, but it doesn't apply here, as that issue is related to unimeter running in ecotracker mode. Like I said, probably just a bit correctly answered request from the Noah, but we need some sniffing to see what is needed exactly. Or maybe run Noah with unimeter in debug mode and check which queries are sent and answered so we can work from there.

oldboys92, 2026-02-09:
have a look at this PR #81 which was confirmed in #80 and also tested by me with a Growatt NEXA 2000.

oldboys92, 2026-03-14:
@TheRealMoeder I believe with all the new improvements a new release make sense?