TheRealMoeder/Energy2Shelly_ESP #44: case sensitive rpc

GalthrikHit, 2025-05-25:
Hello,

I get wrong result, when I use chrome browser in my Android device:
http://shelly_ip/rpc/EM.GetStatus
{"id":1,"src":"shellypro3em-8caab5ef9580","dst":"user_1","result":{"name":"shellypro3em-8caab5ef9580","id":"shellypro3em-8caab5ef9580","mac":"8caab5ef9580","slot":1,"model":"SPEM-003CEBEU","gen":"2","fw_id":"20241011-114455/1.4.4-g6d2a586","ver":"1.4.4","app":"Pro3EM","auth_en":false,"profile":"triphase"}}

same, if in my SW on a ESP device. 
Switching to /rpc/status, everything is Ok.

I can reproduce the behavior in a linux console , when typing:

curl shelly_ip/rpc/em.getstatus 
=> wrong answer

but works properly with case sensitive request
curl shelly_ip/rpc/EM.GetStatus

To my knowledge the HTTP standard is working with lower cases. But I might be wrong.

Would you please be so kind and add lower case answers, too?

TheRealMoeder, 2025-05-27:
I don't have any Shelly devices currently, but will receive a plug next month probably, so I can run tests then.

oldboys92, 2026-02-27:
@GalthrikHit The Shelly API docs provide case sensitive endpoints AFAIK. Let you decide, if you close this issue or not.