TheRealMoeder/Energy2Shelly_ESP #53: Configuration portal ,after WIFI Setup

atomi23, 2025-07-13:
Hello is there any chance to get the config Page after setting Up WIFI , becouse setting up with mobile Phone are ok , but config for mqqtt or http re guest are not very nice on mobile devices ? , and some Protect to reset ? Becouse only easy /reset without any confirmation or password are really risky .

TungstenE2, 2026-08-27:
any update on this request?

I am just setting up new device is the setup process is pain in the b...

GalthrikHit, 2026-09-23:
see version v0.7. WiFi setting, SW update and parameter settings can be reached by "update" button on main page. Sensor will reboot into WiFiManager mode, but within your local network, not in access point mode. Here you can reload the main page with your browser and access WiFiManager to reconfigure.

Close issue?

closed 2026-09-25 by TheRealMoeder