TheRealMoeder/Energy2Shelly_ESP #56: Tasmota -> ESP2Shelly -> Marstek Venus E - Referenz der Ausgangsleistung "fehlgeschlagen"

digit666, 2025-07-29:
Hallo zusammen,

ich habe den mit Energy2Shelly (8266, v0.5.2) simulierten Shelly Pro 3 EM im Marstek Venus E eingebunden. Es werden der Total-Wert und die einzelnen (simulierten) Phasenwerte angezeigt. Leider erfolgt keine Regelung, der Akku ist in der Einstellung"Eigenverbrauch" immer im "Standby".

Beim Shelly Pro 3EM wird in der App unter Referenz der Ausgangsleistung immer die Fehlermeldung "Diagnose fehlgeschlagen, bitte erneut versuchen" angezeigt.

Beim originalen Shelly Pro 3EM wurde in anderen Foren auf ein schlechtes WLAN verwiesen. Ich habe aktuell ca. 80% Signalstärke.

Gibt es irgendwelche Erfahrungen mit Engery2Shelly und diesem Fehler?

Crusher606, 2025-07-31:
Was ist die Datenquelle für Energy2Shelly und was wurde bei den Konfiguration eingestellt?

Werden die Daten auf der /Status Seite korrekt abgefragt und angezeigt?

digit666, 2025-08-01:
Datenquelle ist ein Hichi IR mit Tasmota. Die Verbindung ist vorhanden und funktioniert.
Energy2Shelly gibt wie folgt aus:

{
  "id": 0,
  "a_current": -0.27,
  "a_voltage": 230,
  "a_act_power": -62.66,
  "a_aprt_power": -62.66,
  "a_pf": 1,
  "a_freq": 50,
  "b_current": -0.27,
  "b_voltage": 230,
  "b_act_power": -62.66,
  "b_aprt_power": -62.66,
  "b_pf": 1,
  "b_freq": 50,
  "c_current": -0.27,
  "c_voltage": 230,
  "c_act_power": -62.66,
  "c_aprt_power": -62.66,
  "c_pf": 1,
  "c_freq": 50,
  "total_current": -0.82,
  "total_act_power": -187.98,
  "total_aprt_power": -187.98
}

Die Daten scheinen korrekt zu sein.
Der Shelly UDP-Port ist 1010. Mit der 2220 bekommt der Marstek keine Verbindung.

Die Fehlermeldung ist auf dem dritten Bild zu sehen (rote Markierung). Die Meldung erscheint sowohl bei Auswahl der 3-phasigen als auch bei der 1-phasigen Shelly-Nutzung in der Marstek-App

[image]
[image]
[image]
[image]

Crusher606, 2025-08-02:
Bei Total Power json path steht eine Angabe die es im json von /Status nicht gibt? Wo kommt der Wert power.cum her?

Ich würde da den Wert total_act_power vermuten, das im json vom /Staus zu finden wäre.

Nur eine Vermutung, denn ich habe keinen Hitchi mit Tasmota……

feelfree69, 2026-01-06:
The root cause of this Problem ist simply that the emulator reports 1/3 of the total Power on each Phase.
The Marstek wants to know to which phase it is connected, but fails, since it doesn't see the provided power during setup at a single Phase.

Mastermind2025, 2026-01-08:
i have the same problem with v0.53 and Fronius Smartmeter with HTTP Request.
Is it possible to integrate the workaround described in "[](https://github.com/TheRealMoeder/Energy2Shelly_ESP/issues/69)"

->
"So I tried to select at the MARSTEK App the 'Shelly PRO EM-50' (a single phase meter):
Test with Wireshark: MARSTEK sends per UDP-Multicast (to local IP xxxx.xxx.xxx.255 Port 2223) a "EM1.GetStatus" request.

...I extended the code to answer to this request (a quick and dirty hack - not extensive tested !)
first test: WOOW ! now the zero-energy-mode works :-))"